carla-simulator/scenario_runner #667: RSS Sensor on Scenario Runner

lucasnmatias, 2020-10-19:
From CARLA [documentation](https://carla.readthedocs.io/en/latest/adv_rss/) I've seen that there is a RSS sensor that can be built on CARLA. I would like to know if it would be possible to build and use it on Scenario Runner (I'm using CARLA 0.9.8). If it's possible how I could do that?

fabianoboril, 2020-10-19:
Hi,
there are a couple of options how you can use the sensor.
1: Add the sensor similar to the collision sensor:
https://github.com/carla-simulator/scenario_runner/blob/60d6c1588c07d6b2eef55e95974f403ee52723ca/srunner/scenariomanager/scenarioatomics/atomic_criteria.py#L285
You could add an "RssTest", that makes use of RSS.
2: You could add the sensor directly to your agent (see the CARLA documentation on how to do that).

In general I would recommend you to upgrade to CARLA 0.9.10 if you want to use RSS, as many improvements where made to the sensor and the user interface.

lucasnmatias, 2020-10-19:
Thanks! I managed to attach the sensor to the car but the callback function is never called. I have defined a print callback function to see the data `rss_sensor.listen(lambda rss_data: print(rss_data))` but no data is printed.

DeepAnonymous, 2021-03-03:
Hi @fabianoboril 

According to your saying, one > could add the sensor directly to your agent

In order to do this, in which files we should modify? manual_control.py?

Please help to give an update about this, I hope to add **cameras** and other sensors to see how these sensors record the environment.

Thank you very much.