carla-simulator/scenario_runner #670: AttributeError: module 'carla' has no attribute 'WeatherParameters'

TeeManiac, 2020-10-20:
Build Carla from source (v 0.9.10) and build scenario runner from source (v 0.9.10) in Windows 10 using python 3.9

Added the environment variables and python paths as shown:
set CARLA_ROOT = C:\Users\simple\Desktop\Carla
set SCENARIO_RUNNER_ROOT = C:\Users\simple\Documents\scenario Runner\scenario_runner
set PYTHONPATH=C:\python;C:\Users\simple\Desktop\Carla\PythonAPI\carla\dist\carla-0.9.10-py3.9-win-amd64
set PYTHONPATH=C:\python;C:\Users\simple\Desktop\Carla\PythonAPI\carla
set PYTHONPATH=C:\python;%CARLA_ROOT%\PythonAPI\carla
set PYTHONPATH=C:\python;C:\Users\simple\Desktop\Carla\PythonAPI

Note: In C:\Users\simple\Desktop\Carla\PythonAPI\carla\dist\ .egg filename was 'carla-0.9.10-py3.9-win-amd64' without .egg . So while setting the python path, set the path as:
set PYTHONPATH=C:\python;C:\Users\simple\Desktop\Carla\PythonAPI\carla\dist\carla-0.9.10-py3.9-win-amd64

Then tried the following:

In a new terminal, 
Ran the carla server from carla root directory using the command: make launch
Pressed Play in the UE editor.

In another terminal,
executed: python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld
and got the below error:

C:\Users\simple\Documents\scenario Runner\scenario_runner>python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld                     
Traceback (most recent call last):                                                                                                                        
File "C:\Users\simple\Documents\scenario Runner\scenario_runner\scenario_runner.py", line 35, in <module>                                                 from srunner.scenarioconfigs.openscenario_configuration import OpenScenarioConfiguration                                                              File "C:\Users\simple\Documents\scenario Runner\scenario_runner\srunner\scenarioconfigs\openscenario_configuration.py", line 21, in <module>              
from srunner.scenarioconfigs.scenario_configuration import ActorConfigurationData, ScenarioConfiguration                                              File "C:\Users\simple\Documents\scenario Runner\scenario_runner\srunner\scenarioconfigs\scenario_configuration.py", line 65, in <module>                  
class ScenarioConfiguration(object):                                                                                                                  File "C:\Users\simple\Documents\scenario Runner\scenario_runner\srunner\scenarioconfigs\scenario_configuration.py", line 83, in ScenarioConfiguration     
weather = carla.WeatherParameters()                                                                                                                 
AttributeError: module 'carla' has no attribute 'WeatherParameters' 

**Desktop :**
 - OS: [Windows 10]
 - CARLA Version [CARLA 0.9.10]
 - Python version [3.9.0] [Only python 3.9 is installed in system]
 - Scenario_runner Version [0.9.10]

Please provide support for this issue. Is scenario_runner compatible with python 3.5 and 3.6 only and not with 3.9?

jy723, 2020-10-22:
maybe the python version is not match.
I use carla0.9.10 python3.7

TeeManiac, 2020-10-22:
The issue was not with python 3.9 but with .egg file.

closed 2020-10-22 by TeeManiac

LucasFKobernic, 2021-08-30:
Hi, @TeeManiac 
I'm facing the same problem as you. How did you solve it?

jingyue01, 2023-08-07:
Hi, @TeeManiac @LucasFKobernic 
I'm facing the same problem as you. How did you solve it? I'm using python3.7 and carla-0.9.10-py3.7-win-amd64.egg.