carla-simulator/scenario_runner #672: Leading Vehicle stagnant in scenario

asgardus, 2020-10-26:
Hi,

I invoked the scenario FollowLeadingVehicle_1 and then invoked manual_control.py to have an ego vehicle, which worked..

I drive up behind the spawned vehicle by the scenario, but the spawned vehicle doesn't move and nothing really happens until 60 seconds pass and the scenario times out. The same happened with FollowLeadingVehicleWithObstacle_1. Is this supposed to happen? According to scenario description, I expected spawned actors to have some dynamic behavior.

**Desktop (please complete the following information):**
 - OS: Ubuntu 16.04
 - CARLA Version CARLA 0.9.10 (source)
 - Python version 2.7.17
 - Scenario Runner Version 0.9.10

glopezdiest, 2020-10-27:
that definitely shouldn't happen, we'll take a look

fabianoboril, 2020-10-27:
Hi,
I tested the scenario with Python 2.7 and Ubuntu 16.04 using the current master of SR. For me this works. Could you please run the scenario with the "--debug" option. This should visualize the behavior tree, and would allow us to get an understanding, which node is not properly executed.

asgardus, 2020-10-28:
Hi, thanks for the prompt response. I am away from my system and will get back to you with logs in a day or two, thanks again..!

asgardus, 2020-10-30:
[scenario_bug_report.txt](https://github.com/carla-simulator/scenario_runner/files/5465146/scenario_bug_report.txt)

Output of scenario command
Scenario start
[image]

Scenario end
[image]

glopezdiest, 2020-10-30:
Oh, I see :slightly_smiling_face:. **You have to use ScenarioRunner's manual control, not CARLA's version**.

Regarding CARLA's version of manual control, in order to control a vehicle, it first has to be spawned, and this is done at a random point in the town. This however, isn't required at ScenarioRunner, as we want to control a specific vehicle which has already been spawned. That's why there are two versions.

The black Lincoln you are seeing in front of the police car is, in fact, the ego vehicle of the scenario and when you use SR's manual control, you'll be able to drive it. You'll see that the scenario starts when you start moving.

asgardus, 2020-10-30:
I feel stupid now, thanks, maybe I lazy read the documentation and files with same names are a nightmare for me so I couldn't make the distinction. I'll try this.

closed 2020-10-30 by asgardus

glopezdiest, 2020-11-02:
That's the number one issue users have when starting to use SR, don't worry too much about it :slightly_smiling_face: