carla-simulator/scenario_runner #673: Examples of AssignRouteAction in OpenSCENARIO?

kino3, 2020-10-28:
Are there any examples of using `AssignRouteAction` and `AcquirePositionAction` tag in OpenSCENARIO which have tested already in the latest plugin?
I am trying to use this tag to turn left or right in a cross junction, but I haven't yet.
There are no examples of these tags in the "examples" directory.
https://carla-scenariorunner.readthedocs.io/en/latest/openscenario_support/ saids that these tags are implemented in 0.9.9 (and I checked the source code), so it would be helpful that you provide some example (tested) OpenSCENARIO files.

Lu7913, 2020-10-29:
I did some testing on _RouteAction_, it seems that the vehicle goes straight to the way point, I'm not sure which _RouteStrategy_ ScenarioRunner/Carla supports, if any in terms of openSCENARIO. Since I'm having troubles with _WorldPosition_ (My reference point seems to be wrong, or something), i did this with _LanePosition_.  If you define enough waypoints you can make it work.

This example is vehicle turning in Town04, it works, because of enough waypoints defined. But if you define one waypoint before and one after intersection, vehicle wouldn't perform turning as it should. 
```xml
              <Action name="Assign Route">
                <PrivateAction>
                  <RoutingAction>
                    <AssignRouteAction>
                      <Route name="Route 1" closed="false">
                        <Waypoint routeStrategy="shortest">
                          <Position>
                            <LanePosition roadId="10" s="20.0" laneId="1" offset="0.0">
                              <Orientation h="0.0" p="0.0" r="0.0" type="relative"/>
                            </LanePosition>
                          </Position>
                        </Waypoint>
                        <Waypoint routeStrategy="shortest">
                          <Position>
                            <LanePosition roadId="10" s="15.0" laneId="1" offset="0.0">
                              <Orientation h="0.0" p="0.0" r="0.0" type="relative"/>
                            </LanePosition>
                          </Position>
                        </Waypoint>
                        <Waypoint routeStrategy="shortest">
                          <Position>
                            <LanePosition roadId="10" s="10.0" laneId="1" offset="0.0">
                              <Orientation h="0.0" p="0.0" r="0.0" type="relative"/>
                            </LanePosition>
                          </Position>
                        </Waypoint>
                        <Waypoint routeStrategy="shortest">
                          <Position>
                            <LanePosition roadId="10" s="5.0" laneId="1" offset="0.0">
                              <Orientation h="0.0" p="0.0" r="0.0" type="relative"/>
                            </LanePosition>
                          </Position>
                        </Waypoint>
                        <Waypoint routeStrategy="shortest">
                          <Position>
                            <LanePosition roadId="10" s="0.0" laneId="1" offset="0.0">
                              <Orientation h="0.0" p="0.0" r="0.0" type="relative"/>
                            </LanePosition>
                          </Position>
                        </Waypoint>
                        <Waypoint routeStrategy="shortest">
                          <Position>
                            <LanePosition roadId="1465" s="13.0" laneId="1" offset="1.0">
                              <Orientation h="0.0" p="0.0" r="0.0" type="relative"/>
                            </LanePosition>
                          </Position>
                        </Waypoint>
                        <Waypoint routeStrategy="shortest">
                          <Position>
                            <LanePosition roadId="17" s="56.0" laneId="1" offset="0.0">
                              <Orientation h="0.0" p="0.0" r="0.0" type="relative"/>
                            </LanePosition>
                          </Position>
                        </Waypoint>
                        <Waypoint routeStrategy="shortest">
                          <Position>
                            <LanePosition roadId="17" s="50.0" laneId="1" offset="0.0">
                              <Orientation h="0.0" p="0.0" r="0.0" type="relative"/>
                            </LanePosition>
                          </Position>
                        </Waypoint>
                      </Route>
                    </AssignRouteAction>
                  </RoutingAction>
                </PrivateAction>
              </Action>
```

glopezdiest, 2020-11-04:
Hey, sorry for the delay. As you've pointed out, there are currently no examples of the AssignRouteAction at the repository. 

However, could you explain more in detail what 
> But if you define one waypoint before and one after intersection, vehicle wouldn't perform turning as it should

means? This might be because of the controller but by default, the route waypoints given through the *AssignRouteAction* are used to create a more dense route which takes into account the map lanes, intersections... so the turn should be correct

Lu7913, 2020-11-04:
> means? This might be because of the controller but by default, the route waypoints given through the AssignRouteAction are used to create a more dense route which takes into account the map lanes, intersections... so the turn should be correct

For me this meant that vehicle would go straight to the next waypoint, which would result in cutting corners  or crashing into objects.
As I've just saw in #675, controller must be assigned to the vehicle in order to get this working. 
Sorry, for my misunderstanding.

glopezdiest, 2020-11-04:
Ah don't worry about it, I understand that that part of OpenSCENARIO can easily lead to confusion. All's good as long as it ends up working.

@kino3 I'm closing this issue as @Lu7913 has already provided a working answer. Feel free to open it if you feel like its incomplete, or you have further doubts

fabianoboril, 2020-11-04:
We just added support to use other routing strategies besides "shortest". Please see PR #677

kino3, 2020-11-05:
Thanks for all the comments! I will try again.

kino3, 2020-11-08:
@Lu7913 Hi, could you give me the full (works as it is) example OpenSCENARIO file which you mentioned before? I guess your example was an excerpt. In addition to that, it would be helpful if you could explain the meaning of "enough" waypoint. I didn't understand how do we know our defined waypoints are enough or not.

closed 2020-12-04 by kino3

TigerStone93, 2022-11-01:
@Lu7913 @fabianoboril 
Thank you for your comments. Those are really helpful to me.

jiheddachraoui, 2023-03-24:
hello what type of controller should we assign the simple_vehicle_control dont turn using waypoints

TigerStone93, 2023-03-24:
> @fabianoboril @glopezdiest @TigerStone93 hello what type of controller should we assign the simple_vehicle_control dont turn using waypoints. it goes straight forward and the window crashes at the first curve or intersection

The reason of the malfunction of the controller that you mentioned is hard to explain.

I think it is a kind of bug of the scenario_runner.

See Lu7913's first comment describing the issue related with the intersection . It might be helpful.

jiheddachraoui, 2023-03-24:
yes i defined 16 waypoints i dont know what is considered enough . 
This is an example in Town02 (for routeStrategy I tried them all):
<PrivateAction>
                                    <RoutingAction>
                                        <AssignRouteAction>
                                            <Route name="OSC Generated Route" closed="false">
                                                <Waypoint routeStrategy="random">
                                                    <Position>
                                                        <WorldPosition x="183.8456149001499" y="-191.51437970904" z="-3.1677549169675465e-07" h="-0.00045927497134821933"/>
                                                    </Position>
                                                </Waypoint>
                                                <Waypoint routeStrategy="random">
                                                    <Position>
                                                        <WorldPosition x="184.74605458775707" y="-191.3588298311377" z="-3.1957227961592525e-07" h="-0.0004592837586628025"/>
                                                    </Position>
                                                </Waypoint>
                                                <Waypoint routeStrategy="random">
                                                    <Position>
                                                        <WorldPosition x="185.8030924820124" y="-191.0606925645767" z="-2.7673693208180397e-07" h="0.03814232898574841"/>
                                                    </Position>
                                                </Waypoint>
                                                <Waypoint routeStrategy="random">
                                                    <Position>
                                                        <WorldPosition x="187.16027693875915" y="-190.77551778817403" z="-1.6032676182169442e-07" h="0.14282667884637631"/>
                                                    </Position>
                                                </Waypoint>
                                                <Waypoint routeStrategy="random">
                                                    <Position>
                                                        <WorldPosition x="188.17816528160859" y="-190.24405570354736" z="-1.2215203803679471e-07" h="0.2749703727581613"/>
                                                    </Position>
                                                </Waypoint>
                                                <Waypoint routeStrategy="random">
                                                    <Position>
                                                        <WorldPosition x="188.89590706189816" y="-189.790368558" z="-2.1715510526579235e-07" h="0.32990086982480105"/>
                                                    </Position>
                                                </Waypoint>
                                                <Waypoint routeStrategy="random">
                                                    <Position>
                                                        <WorldPosition x="189.4309509345025" y="-189.42741884152693" z="-2.0784839714132812e-07" h="0.40711406759254504"/>
                                                    </Position>
                                                </Waypoint>
                                                <Waypoint routeStrategy="random">
                                                    <Position>
                                                        <WorldPosition x="191.03608255201905" y="-188.6885569191649" z="-1.920290581063514e-07" h="0.7728122187044748"/>
                                                    </Position>
                                                </Waypoint>
                                                <Waypoint routeStrategy="random">
                                                    <Position>
                                                        <WorldPosition x="191.7538243322452" y="-188.325607202854" z="-2.711981975310118e-07" h="0.9626605799183507"/>
                                                    </Position>
                                                </Waypoint>
                                                <Waypoint routeStrategy="random">
                                                    <Position>
                                                        <WorldPosition x="192.47156611282333" y="-187.573782789807" z="-2.9821977879998834e-07" h="1.1364782470225747"/>
                                                    </Position>
                                                </Waypoint>
                                                <Waypoint routeStrategy="random">
                                                    <Position>
                                                        <WorldPosition x="192.954410583732" y="-186.69233347763543" z="-3.1127391668649953e-07" h="1.3102959141419193"/>
                                                    </Position>
                                                </Waypoint>
                                                <Waypoint routeStrategy="random">
                                                    <Position>
[... cut ...]
                                                        <WorldPosition x="193.69825207382283" y="-176.2445666259376" z="-5.517546670486783e-06" h="1.5710224148330088"/>
                                                    </Position>
                                                </Waypoint>
                                            </Route>
                                        </AssignRouteAction>
                                    </RoutingAction>
                                </PrivateAction>

TigerStone93, 2023-03-25:
> @TigerStone93 yes i saw @Lu7913 comment . I defined 16 waypoints i dont know what is considered enough . This is an example in Town02 (for routeStrategy I tried them all):
> 
> > 
> 
> ```
>                                 <RoutingAction>
>                                     <AssignRouteAction>
>                                         <Route name="OSC Generated Route" closed="false">
>                                             <Waypoint routeStrategy="random">
>                                                 <Position>
>                                                     <WorldPosition x="183.8456149001499" y="-191.51437970904" z="-3.1677549169675465e-07" h="-0.00045927497134821933"/>
>                                                 </Position>
>                                             </Waypoint>
>                                             <Waypoint routeStrategy="random">
>                                                 <Position>
>                                                     <WorldPosition x="184.74605458775707" y="-191.3588298311377" z="-3.1957227961592525e-07" h="-0.0004592837586628025"/>
>                                                 </Position>
>                                             </Waypoint>
>                                             <Waypoint routeStrategy="random">
>                                                 <Position>
>                                                     <WorldPosition x="185.8030924820124" y="-191.0606925645767" z="-2.7673693208180397e-07" h="0.03814232898574841"/>
>                                                 </Position>
>                                             </Waypoint>
>                                             <Waypoint routeStrategy="random">
>                                                 <Position>
>                                                     <WorldPosition x="187.16027693875915" y="-190.77551778817403" z="-1.6032676182169442e-07" h="0.14282667884637631"/>
>                                                 </Position>
>                                             </Waypoint>
>                                             <Waypoint routeStrategy="random">
>                                                 <Position>
>                                                     <WorldPosition x="188.17816528160859" y="-190.24405570354736" z="-1.2215203803679471e-07" h="0.2749703727581613"/>
>                                                 </Position>
>                                             </Waypoint>
>                                             <Waypoint routeStrategy="random">
>                                                 <Position>
>                                                     <WorldPosition x="188.89590706189816" y="-189.790368558" z="-2.1715510526579235e-07" h="0.32990086982480105"/>
>                                                 </Position>
>                                             </Waypoint>
>                                             <Waypoint routeStrategy="random">
>                                                 <Position>
>                                                     <WorldPosition x="189.4309509345025" y="-189.42741884152693" z="-2.0784839714132812e-07" h="0.40711406759254504"/>
>                                                 </Position>
>                                             </Waypoint>
>                                             <Waypoint routeStrategy="random">
>                                                 <Position>
>                                                     <WorldPosition x="191.03608255201905" y="-188.6885569191649" z="-1.920290581063514e-07" h="0.7728122187044748"/>
>                                                 </Position>
>                                             </Waypoint>
>                                             <Waypoint routeStrategy="random">
>                                                 <Position>
>                                                     <WorldPosition x="191.7538243322452" y="-188.325607202854" z="-2.711981975310118e-07" h="0.9626605799183507"/>
>                                                 </Position>
>                                             </Waypoint>
>                                             <Waypoint routeStrategy="random">
>                                                 <Position>
>                                                     <WorldPosition x="192.47156611282333" y="-187.573782789807" z="-2.9821977879998834e-07" h="1.1364782470225747"/>
>                                                 </Position>
>                                             </Waypoint>
>                                             <Waypoint routeStrategy="random">
>                                                 <Position>
>                                                     <WorldPosition x="192.954410583732" y="-186.69233347763543" z="-3.1127391668649953e-07" h="1.3102959141419193"/>
>                                                 </Position>
>                                             </Waypoint>
>                                             <Waypoint routeStrategy="random">
>                                                 <Position>
>                                                     <WorldPosition x="193.3850556530416" y="-185.16275967085176" z="-3.150251867673644e-07" h="1.5202463916416047"/>
>                                                 </Position>
[... cut ...]
>                                         </Route>
>                                     </AssignRouteAction>
>                                 </RoutingAction>
>                             </PrivateAction>
> ```

I do not know why but the `WorldPosition` did not work properly in my case, and showed a similar bug that you have explained.

I would recommend you to use `LanePosition` instead.

However, maybe you should find the `roadId` what you want one by one by yourself because the `roadId` is not displayed anywhere in the CARLA GUI.

jiheddachraoui, 2023-03-27:
@TigerStone93 sadly its the same problem while using LanePosition i think the example that @Lu7913 provided is an except . @Lu7913 can you provide the full code?