carla-simulator/scenario_runner #675: Any alternative way to maneuver vehicles in curved road

vikilancer, 2020-11-02:
I am trying to maneuver external actors (traffic vehicles) in a 2000m long curved road. 
Currently I am using a tag named "AssignRouteAction" and  have manually provided the worldposition(x,y) to move the vehicle in the curved path.
Is there any other possible way to maneuver the external actors in curved road without providing worldposition in .xosc?

[image]

 - OS: [e.g. Ubuntu 18.04]
 - CARLA Version [e.g. CARLA 0.9.9]
 - Python version [e.g. 3.6]

glopezdiest, 2020-11-04:
Hey @vikilancer, there are basically two options to move a vehicle:

The first is one is to use **CARLA's autopilot**, which you can set by *ActivateControllerAction*. With that the **vehicle will randomly move around the town.** The main disadvantage is that you can't it's route so at intersections, the actor will choose a random exit.

The second one is to use an ***AssignRouteAction*** (which you are already using) **+ a controller**. **Waypoints can be as far away as you want** as the route between them is interpolated, so you only need to specify the important ones (such as the waypoints after an intersection).

Additionally, [here](https://carla-scenariorunner.readthedocs.io/en/latest/openscenario_support/#level-of-support), you have the level of support of OpenSCENARIO 1.0, and at the end, all the types of positions you can use.

fabianoboril, 2020-11-04:
> The second one is to use an _**AssignRouteAction**_ (which you are already using) **+ a controller**. **Waypoints can be as far away as you want** as the route between them is interpolated, so you only need to specify the important ones (such as the waypoints after an intersection).

This is only partially correct. For the shortest routingStrategy this does not happen. There really a direct connection is used. However, by a real coincidence, I required this feature today, so there is now a PR for this #673

closed 2021-11-23 by glopezdiest