carla-simulator/scenario_runner #678: TTC for MiscObject error

sagar-g-v, 2020-11-09:
Hi,
I am trying to use TTC with respect to static object (barrier) and vehicle.

but i am geeting the bellow error.

[image]

It is saying that static object does not have bounding_box attribute.

I think this bug is introduced in #666 when freespace is implemented.

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04
 - CARLA Version 0.9.10.1
 - Python version 3.7
 - Version 0.9.10

glopezdiest, 2020-11-09:
hey @sagar-g-v. I'll take a look at it. Could you pass me the .xodr you are using in order for me to reproduce it?

sagar-g-v, 2020-11-09:
@glopezdiest 

[Test.zip](https://github.com/carla-simulator/scenario_runner/files/5509923/Test.zip)

attached .xosc file to reproduce the above issue
found the Issue to be because of barrier which does not have `bounding_box` attribute and made a small change in my local file 

`InTimeToArrivalToVehicle`
```python
def update(self):
        """
        Check if the ego vehicle can arrive at other actor within time
        """
        new_status = py_trees.common.Status.RUNNING

        current_location = CarlaDataProvider.get_location(self._actor)
        other_location = CarlaDataProvider.get_location(self._other_actor)

        is_misc_obj = (str(self._actor.type_id).startswith('static') or
                       str(self._other_actor.type_id).startswith('static'))

        if current_location is None or other_location is None:
            return new_status

        current_velocity = CarlaDataProvider.get_velocity(self._actor)
        other_velocity = CarlaDataProvider.get_velocity(self._other_actor)

        if self._along_route:
            # Global planner needs a location at a driving lane
            current_location = self._map.get_waypoint(current_location).transform.location
            other_location = self._map.get_waypoint(other_location).transform.location

        distance = calculate_distance(current_location, other_location, self._grp)

        # if velocity is too small, simply use a large time to arrival
        time_to_arrival = self._max_time_to_arrival

        if current_velocity > other_velocity:
            if self._condition_freespace and not is_misc_obj:
                time_to_arrival = (distance - self._actor.bounding_box.extent.x -
                                   self._other_actor.bounding_box.extent.x) / (current_velocity - other_velocity)
            else:
                time_to_arrival = distance / (current_velocity - other_velocity)

        if self._comparison_operator(time_to_arrival, self._time):
            new_status = py_trees.common.Status.SUCCESS

        self.logger.debug("%s.update()[%s->%s]" % (self.__class__.__name__, self.status, new_status))

        return new_status
````
Hope this will resuce your effort in finding the cause.

Thank you.

glopezdiest, 2020-11-09:
Thanks, I'll take a look ASAP

glopezdiest, 2020-11-09:
Yeah that's definitely an issue. It isn't related to any PR though. The problem is that currently, *carla.Actor*, have any bounding box attributes, whereas vehicles and walkers do.

PR #679 (2020-11-18): Fixed InTimeToArrivalToVehicle bug with static objects
### Description

Currently, while vehicles and walkers have bounding boxes, all other objects such as static props or traffic signs don't. This was causing an exception when InTimeToArrivalToVehicle was being used to check for such objects. An additional condition has been placed, to avoid such exception.

Fixes #678

### Where has this been tested?

  * **Platform(s):** Ubutnu 18.04
  * **Python version(s):** 3.6
  * **Unreal Engine version(s):** 4.24
  * **CARLA version:** 0.9.10.1

---
This change is [[image]](https://reviewable.io/revi [... cut]
closed 2020-11-18 by fabianoboril