carla-simulator/scenario_runner #682: No output file generated for openScenario

TeeManiac, 2020-11-10:
While running openScenario based scenarios in scenario runner, the result files are not getting generated. Tried the following commands:
python scenario_runner.py --openscenario srunner\examples\LaneChangeSimple.xosc  --reloadWorld --output --json and also tried
python scenario_runner.py --openscenario srunner\examples\LaneChangeSimple.xosc  --reloadWorld --output --file

A file named CARLA of size 0kb is getting created everytime I try with these commands. The result files are getting generated for basic scenarios but not for openscenario. Is result file not supported for openScenario?

As shown in the screenshot, the json files are generated foe basic_scenario whereas the file named CARLA is getting generated for openscenario execution.

**Desktop
[image]
:**

OS: [Windows 10]
CARLA Version [CARLA 0.9.10]
Python version [3.9.0] [Only python 3.9 is installed in system]
Scenario_runner Version [0.9.10]

glopezdiest, 2020-11-18:
Hey @TeeManiac, sorry for the delay. Checking the issues I thought this one was already solved. Did you manage to solve it? I just checked and it's working fine for me. A file named *CARLA:CyclistCrossing2020-11-18-09-57-58.txt* was created, with all the info needed.

However, I am working on Ubuntu, and not Windows so that might be an issue (maybe the *:* have a specific meaning there?). If you haven't solved it yet please try the following

1. Change the starting name of the OpenScenario file from *CARLA:* to *CARLA_* (1st line of the *.xosc*)
2. Go to *srunner/scenarioconfigs/openscenario_configurations.py and change the check at *_set_scenario_name()* for *CARLA_*

[image]
 
Let me know if this fixes it, because we might want to change that part if it is a Windows related problem

TeeManiac, 2020-11-19:
> Hey @TeeManiac, sorry for the delay. Checking the issues I thought this one was already solved. Did you manage to solve it? I just checked and it's working fine for me. A file named [...]

I fixed it by replacing ":" with "" in config_name and its working now.

fabianoboril, 2020-11-19:
@TeeManiac Thanks for this info. Maybe we should just remove this "CARLA:" for the output file, as it is more or less just an internal way of handling the different coordinate systems.

TeeManiac, 2020-11-19:
> @TeeManiac Thanks for this info. Maybe we should just remove this "CARLA:" for the output file, as it is more or less just an internal way of handling the different coordinate systems.

@fabianoboril Yes, I too think so as the json file for basic scenario just has the Scenario name + current time for the json file name.

PR #691 (2020-11-26): Fixed writing result files with OSC under Windows
Using the "CARLA:" prefix in the OSC description caused broken result files
under Windows. Therefore, the internal handling of the scenario name was modified
to enable writing the output files under Windows and Linux.

Fixes #682

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/scenario_runner/691)
closed 2020-11-26 by fpasch