carla-simulator/scenario_runner #683: Parameterize weather in scenario runner

TeeManiac, 2020-11-12:
I tried modifying the weather parameters in xml file for basic scenario configuration. But the weather parameters are not getting reflected in the carla world while running the scenario. But using manual control (Ctrl+C), I am able to visualize all the available weather conditions in the scenario. How can we parameterize the weather conditions in scenario runner so as to visualize the parameter changes for basic_scenario configuration?

**Desktop **

OS: [Windows 10]
CARLA Version [CARLA 0.9.10]
Python version [3.9.0]
Scenario_runner Version [0.9.10]

glopezdiest, 2020-11-16:
Hey @TeeManiac.
> I tried modifying the weather parameters in xml file for basic scenario configuration

What exactly do you mean here? have you modified the xml like the example below?
[image]

I just tested it and its working fine on my side

TeeManiac, 2020-11-17:
@glopezdiest I had modified the parameter values as you have mentioned above.

I figured out the issue. I had a scenario runner release downloaded in the same working dir as Carla. But it had some issues. So I was working on scenario runner built from source in another directory. All the paths were set according to the new working directory. But scenario_parser was by default taking the cwd for scenarios. 

Also, its mentioned in the documentations that the working directory of Carla and scenario runner does not matter. 

Now I have changed the location of scenario runner buit from source to working directory of Carla and set the paths accordingly and its working fine.

glopezdiest, 2020-11-17:
Good to know its fixed

> Also, its mentioned in the documentations that the working directory of Carla and scenario runner does not matter.

As far as I know, it doesn't, and all the tests I've done haven't given me any problems. ScenarioRunner knows where CARLA's API is with the use of the PYTHONPATH environment variable, so as long as that one is correct, the specific folder of SR is not important.

closed 2020-11-18 by glopezdiest