carla-simulator/scenario_runner #684: Creating new scenario .

akshatamk, 2020-11-17:
Hello,
   how to create new scenarios in carla. Is it only only by creating python file(scripting) and the related XML file? Is there any scenario editor available for carla?
Please help!

seowwj, 2020-11-18:
Related to #661

glopezdiest, 2020-11-18:
Currently, there is no editor available for the creation of scenarios. As you pointed out you have to either use the python file, or use the OpenScenario format.

akshatamk, 2020-11-25:
@glopezdiest Thank you for the response

GeneralMak, 2021-07-23:
I have the similar problem,the official said to create a new scenario (https://carla-scenariorunner.readthedocs.io/en/latest/creating_new_scenario.html),four methods need to be defined .But how to define the parameters of this  _initialize_actors method  and create_behavior method, I have'nt found a good turorial.  Ask for help!!!

GeneralMak, 2021-07-23:
I imitated the official examples to define the parameter values in my own method,but various errors will be reported during runtime,so I would like to ask for if any scholars have made them.

GeneralMak, 2021-07-23:
@fabianoboril  hello, Can you guys give a detailed tutorial on how to create Initialize Method and CreateBehavior method?(https://carla-scenariorunner.readthedocs.io/en/latest/creating_new_scenario.html)

fabianoboril, 2021-07-23:
Why not using OpenSCENARIO? This is much easier, and there is a graphical scenario editor: https://github.com/carla-simulator/traffic-generation-editor?

GeneralMak, 2021-07-26:
@fabianoboril  ok, Thank you for the response! I'will try.

closed 2021-11-23 by glopezdiest