carla-simulator/scenario_runner #688: Agent support for all scenarios in one class

TeeManiac, 2020-11-19:
Currently, when trying to run all the scenarios of one class, manual_control.py needs to be run separately whenever a scenario in the class is prepared. Is there any existing methods to keep the agent wait for ego vehicle while the other scenarios are getting prepared? If no, can you please add this feature?

OS: [Windows 10]
CARLA Version [CARLA 0.9.10]
Python version [3.9.0]
Scenario_runner Version [0.9.10]

glopezdiest, 2020-11-19:
That's currently not supported as right now the manual control automatically shuts down when the ego is destroyed. In order to add your feature, some kind of reset should have to be implemented, but I don't think it is as easy as it sounds. I'm sorry but there are no plans to do so right now, but feel free to do so and do a PR :slightly_smiling_face:

TeeManiac, 2020-11-19:
> That's currently not supported as right now the manual control automatically shuts down when the ego is destroyed. In order to add your feature, some kind of reset should have to be implemented, but [...]

@glopezdiest  Thank you for the response. Will do a PR for this and will also try this from my side.

closed 2021-11-23 by glopezdiest