carla-simulator/scenario_runner #694: Automatic Control

SiNa-Dehbari, 2020-11-24:
Hello Everybody,

I have a basic question.

I want to run scenarios (Ego vehicles and actors) automatically. It means they move base on  waypoints or any other method, by their own. In other words, I don't want to use `manual_control.py`

could you please let me now **which controller** and **how** should I use?

Thank you in advance

fabianoboril, 2020-11-26:
There are a couple of options:
1.	You run manual_control.py with autopilot enabled, but still you have to start manual_control.py then separately.
a.	You can incorporate the start of manual_control.py into the scenario runner, to make it easier to use.
2.	If you use OpenSCENARIO you can make use of a controller to control the ego vehicle. There is an example on github. If you use the simple_vehicle_control, you can even attach a camera to visualize the scenario.
3.	If you use a python-encoded scenario, you can simply enforce actions for the ego vehicle, like you do for any other vehicle. However, you have to attach a camera to the ego vehicle in this case, to see what is happening.

DeepAnonymous, 2021-03-07:
Hi @fabianoboril ,

Could you please help to indicate where is the example with "simple_vehicle_control" on github. I couldn't find it.

I tried to set as follow, but nothing happens. And how to save the attach_camera image as well.
```
<Controller name="HeroAgent">
        <Properties>
               <Property name="module" value="simple_vehicle_control"/>
        </Properties>
</Controller>
```

Thank you.

DeepAnonymous, 2021-03-07:
And how to start / view the scenario as well, when we don't need to run 'manual_control.py' anymore.

Thank you.

Hechao-1116, 2021-06-03:
> 嗨@fabianoboril，
> 
> 您能否帮助指出github上带有“simple_vehicle_control”的示例在哪里。我找不到它。
> 
> 我试图设置如下，但没有任何反应。以及如何为此设置参数，例如速度或 st 以使其运行。
> 
> ```
> <Controller name="HeroAgent">
>         <Properties>
>                <Property name="module" value="simple_vehicle_control"/>
>         </Properties>
> </Controller>
> ```
> 
> 谢谢你。

8/5000 
I have the same problem,and how can I add my own module in the value

werewolfdev, 2021-09-21:
Hi @SiNa-Dehbari  Where you able to run the ego vehicle through a defined route with openSCENARIO using simple_vehicle_control as mentioned above. Iam not able to do the same actually

werewolfdev, 2021-09-21:
@DeepAnonymous https://github.com/carla-simulator/scenario_runner/blob/05a811a6248c8033e4c76185ff40ae5eccad7e69/srunner/examples/OscControllerExample.xosc#L77

TigerStone93, 2022-10-30:
I think the `manual_control.py` itself only can activate the CARLA autopilot, and it cannot utilize the simple_vehicle_control.