carla-simulator/scenario_runner #695: OpenScenario 1.0.0

Vela-zz, 2020-11-25:
Recently, I used esmini/pyoscx  to generate openScenario 1.0.0 standard file, then using carla/scenario_runner collaboration for 
simulating, and I found some Condition tag, such like **TakeHeadAwayCondition**.. and some attribute in tag, like /**freespace**/ in **RelativeDistanceCondition**.

can scenario_runner team add a list about  tags or attributes of openScenario which can be used in carla scenario_runner?

glopezdiest, 2020-11-30:
Hey @Vela-zz, you can check the current support of OSC [here](https://carla-scenariorunner.readthedocs.io/en/latest/openscenario_support/)

Vela-zz, 2020-11-30:
> Hey @Vela-zz, you can check the current support of OSC [here](https://carla-scenariorunner.readthedocs.io/en/latest/openscenario_support/)

Thank you !

closed 2020-11-30 by Vela-zz