carla-simulator/scenario_runner #696: Adding a vehicle in Unreal Editor

tecena, 2020-11-25:
Hello,
I'm new to carla.
I'm modifying XML config file and trying to add (x,y,z) configuration value to place the ego vehicle and other vehicles in the map. Its really taking lot of time in trial and error to place the actors at positions and do not feel it as convenient way.
I have read in one solution as we can open our map in Unreal editor and place the vehicles at the required positions and then place the configuration values in XML file, I tried to do it but could not find the way.
Please suggest as how to open any one of the map given by carla in UE editor and place the actors at the positions.

fabianoboril, 2020-11-26:
For the Unreal editor @glopezdiest might be the best to answer. However, you can also use any OpenDRIVE viewer, get the x-y coordinates, invert the sign of y, and you have your coordinates.

glopezdiest, 2020-11-30:
Hey @tecena. Are you working with the packages or from source? In order to open the CARLA maps with Unreal you have to build CARLA from source. From there, you do a *make launch* to run Unreal which will open up the default town .

[image]

On the left you can see an *Empty Actor*. You can drag into the scene to create that object. On the right, you'll see its transform (in cm, not meters!).

tecena, 2020-12-04:
@fabianoboril @glopezdiest 
Thank you,
I'm using the Carla from source.

closed 2021-11-23 by glopezdiest