carla-simulator/scenario_runner #697: ModuleNotFoundError: No module named 'srunner'

pramo23, 2020-11-27:
**Describe our question or idea**
A clear and concise description.

Hi everyone,
I have installed scenario runner with corresponding carla version. As provided in getting started page, i also exported corresponding path in my environment variable. Please refer to image below.

[image]

I'm able to run 
`python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld`
and `python manual_control.py`
It runs without any problem 
but when i try to run any particular python program from scenario_runner\srunner\scenarios then it gives me error ModuleNotFoundError

λ py control_loss.py
Traceback (most recent call last):
  File "control_loss.py", line 19, in <module>
    from srunner.scenariomanager.carla_data_provider import CarlaDataProvider
ModuleNotFoundError: No module named 'srunner'

Have anyone faced similar problem ?

**Desktop (please complete the following information):**
 - OS: Windows 10 
 - CARLA Version 0.9.9.4
 - Python version 3.7

glopezdiest, 2020-11-30:
Hey @pramo23. Yes, that's probably because most of the ScenarioRunner imports (if not all) are relative to the main scenario_runner folder, so you'll have problems

glopezdiest, 2021-11-23:
@werewolfdev Did you manage to solve the issue? I'm preemptively closing issue as it is a bit old, but feel free to reopen it

closed 2021-11-23 by glopezdiest

werewolfdev, 2021-11-23:
Solved .Thanks