carla-simulator/scenario_runner #700: LaneChangeSimple.xosc Example

TKeutgens, 2020-12-04:
**Describe the bug**
When running the LaneChangeSimple.xosc example on my Windows machine I get the following bug.

`WARNING:[SR:OpenScenarioConfiguration]: Wrong OpenDRIVE map in use. Forcing reload of CARLA world
Preparing scenario: CARLA:LaneChangeSimple
ScenarioManager: Running scenario OpenScenario

Traceback (most recent call last):
  File "H:/11_Carla_Simulator/scenario_runner_git/scenario_runner/scenario_runner.py", line 400, in _load_and_run_scenario
    self.manager.run_scenario()
  File "H:\11_Carla_Simulator\scenario_runner_git\scenario_runner\srunner\scenariomanager\scenario_manager.py", line 134, in run_scenario
    self._tick_scenario(timestamp)
  File "H:\11_Carla_Simulator\scenario_runner_git\scenario_runner\srunner\scenariomanager\scenario_manager.py", line 176, in _tick_scenario
    self.scenario_tree.tick_once()
  File "C:\Users\keutgens\AppData\Local\Programs\Python\Python37\lib\site-packages\py_trees\behaviour.py", line 158, in tick_once
    for unused in self.tick():
  File "C:\Users\keutgens\AppData\Local\Programs\Python\Python37\lib\site-packages\py_trees\composites.py", line 578, in tick
    for node in child.tick():
  File "C:\Users\keutgens\AppData\Local\Programs\Python\Python37\lib\site-packages\py_trees\composites.py", line 578, in tick
    for node in child.tick():
  File "C:\Users\keutgens\AppData\Local\Programs\Python\Python37\lib\site-packages\py_trees\composites.py", line 578, in tick
    for node in child.tick():
  File "C:\Users\keutgens\AppData\Local\Programs\Python\Python37\lib\site-packages\py_trees\behaviour.py", line 249, in tick
    new_status = self.update()
  File "H:\11_Carla_Simulator\scenario_runner_git\scenario_runner\srunner\scenariomanager\scenarioatomics\atomic_criteria.py", line 829, in update
    current_wp = self._map.get_waypoint(current_loc, lane_type=carla.LaneType.Any)
OverflowError: can't convert negative value to unsigned int`

When I print the internals of the the two variable I get the following:
current_loc = Location(x=-9.400000, y=-152.800003, z=0.487832)
lane_type = Any

**Desktop :**
 - OS: Windows 10
 - CARLA Version 0.9.10
 - Python version 3.7
- py-trees 0.8.3

TKeutgens, 2020-12-04:
When switching to the newest py-trees version (2.1.2), the error changes into the following:

```
type object 'ParallelPolicy' has no attribute 'SUCCESS_ON_ALL'
  File "H:/11_Carla_Simulator/scenario_runner_git/scenario_runner/scenario_runner.py", line 373, in _load_and_run_scenario
    timeout=100000)
  File "H:\11_Carla_Simulator\scenario_runner_git\scenario_runner\srunner\scenarios\open_scenario.py", line 188, in __init__
    terminate_on_failure=False, criteria_enable=criteria_enable)
  File "H:\11_Carla_Simulator\scenario_runner_git\scenario_runner\srunner\scenarios\basic_scenario.py", line 63, in __init__
    behavior = self._create_behavior()
  File "H:\11_Carla_Simulator\scenario_runner_git\scenario_runner\srunner\scenarios\open_scenario.py", line 251, in _create_behavior
    policy=py_trees.common.ParallelPolicy.SUCCESS_ON_ALL, name="Story")
AttributeError: type object 'ParallelPolicy' has no attribute 'SUCCESS_ON_ALL'
```

So I guess the issue is related to py-tress.

fabianoboril, 2020-12-04:
> When switching to the newest py-trees version (2.1.2), the error changes into the following:
>
> ```
> type object 'ParallelPolicy' has no attribute 'SUCCESS_ON_ALL'
>   File [...]

Newer py_trees versions are _NOT_ supported.

I will look into the other problem and get back to you.

fabianoboril, 2020-12-04:
Hi,
at the moment I cannot reproduce your Problem with Python 2.7 or 3.5. Are you using CARLA 0.9.10 or 0.9.10.1?

glopezdiest, 2020-12-04:
Hey, that's actually a CARLA related problem for Windows. As of 0.9.10.1, the *get_waypoint()* function fails when used with the *carla.LaneType.Any*. 

These is already fixed on CARLA's *dev* branch and will be updated to master along with the next version update. In the meantime, you can cherry-pick these two commits ([1](https://github.com/carla-simulator/carla/commit/530f56e78b999550670c7e5002b9b740218b8a58)) ([2](https://github.com/carla-simulator/carla/commit/aedd70911905d1cb80844c55cfc324469de07c90)).

If that is impossible, you can go to that file (*srunner/scenariomanager/scenarioatomics/atomic_criteria.py*), and change that line (*829*), removing the *carla.LaneType.Any*:

from `self._map.get_waypoint(current_loc, lane_type=carla.LaneType.Any)`
to `self._map.get_waypoint(current_loc)`

That will make the criteria fail to detect if the vehicle has invaded a sidewalk, but if you don't care about that, there shouldn't be an issue.

PD: as @fabianoboril said, newer py_trees version are not supported, you should go back to version 0.8.3

TKeutgens, 2020-12-04:
@fabianoboril and @glopezdiest thank you very much for your help! The now code line fixed the issue.

closed 2020-12-04 by TKeutgens

TKeutgens, 2021-02-18:
@glopezdiest is the bug already fixed in the version 0.9.11?

glopezdiest, 2021-02-18:
Yeah it is