carla-simulator/scenario_runner #701: Issue: generate_target_waypoint_list_multilane

TKeutgens, 2020-12-04:
First of all thank you for your great work and the great support! 

I tried to edit the LaneChangeSimple.xosc example to a new location. What I discovered was, that the lane change only works if I edit the generate_target_waypoint_list_multilane  function. I had to replace the return to a break in order for the vehicle to execute the lane change.

```
# Other lane
    distance = 0
    while distance < distance_other_lane:
        next_wps = plan[-1][0].next(step_distance)
        if not next_wps:
            break
            #return None, None
        next_wp = next_wps[0]
        distance += next_wp.transform.location.distance(plan[-1][0].transform.location)
        plan.append((next_wp, RoadOption.LANEFOLLOW))
```
I attached the files to reproduce the error. 
[examples.zip](https://github.com/carla-simulator/scenario_runner/files/5644237/examples.zip)

I'm not sure why the lane change does not work properly for this location. 

**Desktop:**
 - OS: Windows 10
 - CARLA Version 0.9.10
- SCENARIORUNNER Version: checkout at commit 11f6788536668e3bcf2ba14cdb25bd24e4c7a8d5
 - Python version 3.7

TKeutgens, 2021-02-18:
Update: Issue still exists for ScenarioRunner 0.9.11

glopezdiest, 2021-11-23:
@TKeutgens Did you manage to solve the issue? I'm preemptively closing issue as it is a bit old, but feel free to reopen it

closed 2021-11-23 by glopezdiest