carla-simulator/scenario_runner #703: OpenScenario support - TeleportAction

Sumza, 2020-12-08:
I have observed that when **TeleportAction** is used, the entity gets teleported to the required location but with the reset in velocity   ( i.e. Entity velocity comes back to zero) and then the consecutive events are executed.

Is this the desired behaviour or any future updations is in progress for the **TeleportAction** tag ?

glopezdiest, 2020-12-11:
Hey @Sumza, we have no current plans to change the TeleportAction, and yes, the reset of velocity is intentional, for stability purposes. Is it bothering you in any way that reset? we are mainly using it when spawning actors, so if that's the case, I'm curious to see your use case

Sumza, 2020-12-11:
Hey @glopezdiest , we are using Teleport action in story to achieve Re-Initialization of traffic objects with desired relative position and keep the velocity. We also observe that Teleport Action affects the stability of TV if started with previously achieved velocity in curve road profile.
Is there any other way to achieve the re-initialization of traffic without reset of velocity during the simulation?

glopezdiest, 2020-12-11:
I see. If you really don't want to reset the velocities, you can directly remove it. You can go to *srunner/scenariomanager/scenarioatomics/atomic_behaviors.py*, class *ActorTransformSetterToOSCPosition* and remove this part of the code at the *initialize* function

```
if self._actor.is_alive:
    self._actor.set_target_velocity(carla.Vector3D(0, 0, 0))
    self._actor.set_target_angular_velocity(carla.Vector3D(0, 0, 0))
```

Sumza, 2021-02-22:
Thankyou for your Input, It helped my requirement.

I tried implementing the Teleport Action on a curved road and below is the observed behaviour video output :

https://user-images.githubusercontent.com/68461250/108724291-2e9d7480-754b-11eb-8666-fa8d701046a4.mp4

When the Traffic vehicle is teleported from one part of the curved road to another part of the same curved road, we can see that the vehicle takes a small time gap to adjust to the teleported part of the curved road.
 

Is there any way this time delay can be avoided while Teleporting vehicles on Curved Road ?

glopezdiest, 2021-02-24:
I'm sorry but for me, it seems like your video is corrupted. Is it also happening on your end? Can you perhaps re-upload it

Sumza, 2021-02-25:
The video is playing fine at my end, i am re-uploading the video.

Kindly check and let me know its running fine you this time.

https://user-images.githubusercontent.com/68461250/109111403-d5a62a00-775e-11eb-86b4-8c58e1d59505.mp4

glopezdiest, 2021-02-25:
Okay, yeah, its works fine for me now. I don't think we ever designed the TeleportAction to work as you are using it so there are bound to be some errors. How are you controlling that vehicle? XOSC controllers use the local planner, which has a waypoint buffer and you probably aren't resetting it, maybe that's causing some issues

WindTi, 2021-12-07:
> Okay, yeah, its works fine for me now. I don't think we ever designed the TeleportAction to work as you are using it so there are bound to be some errors. How are you controlling that vehicle? XOSC [...]

Hello, you mentioned local planner and waypoint buffer, how can I reset them?

glopezdiest, 2021-12-09:
What version of CARLA are you using? We improved the agents a lot for 0.9.12 and they work differently.

For > 0.9.12, check [this variable](https://github.com/carla-simulator/carla/blob/master/PythonAPI/carla/agents/navigation/local_planner.py#L65), which is a list of future waypoints the vehicle will pas through. On teleport, you'd have to reset that to be empty. For previous versions, you'll want to reset two variables ([this](https://github.com/carla-simulator/carla/blob/0.9.11/PythonAPI/carla/agents/navigation/local_planner.py#L76) and [this](https://github.com/carla-simulator/carla/blob/0.9.11/PythonAPI/carla/agents/navigation/local_planner.py#L74)). 

However, I don't know if that alone will be enough as I've personally never tried it.

WindTi, 2021-12-10:
> 您使用的是哪个版本的 CARLA？我们在 0.9.12 中对代理进行了大量改进，它们的工作方式有所不同。
> 
> 对于 > 0.9.12，检查[这个变量](https://github.com/carla-simulator/carla/blob/master/PythonAPI/carla/agents/navigation/local_planner.py#L65)，它是车辆将通过的未来航点列表。在传送时，您必须将其重置为空。对于以前的版本，您需要重置两个变量（[this](https://github.com/carla-simulator/carla/blob/0.9.11/PythonAPI/carla/agents/navigation/local_planner.py#L76)和[this](https://github.com/carla-simulator/carla/blob/0.9.11/PythonAPI/carla/agents/navigation/local_planner.py#L74)）。
> 
> 但是，我不知道仅凭这一点是否足够，因为我个人从未尝试过。

> What version of CARLA are you using? We improved the agents a lot for 0.9.12 and they work differently.
>
> For > 0.9.12, check [this [...]

Thanks for the reply, I am using carla0.9.12. I have not used the method you mentioned, I will try it later. At present, after trying to use RoutingAction->AssignRouteAction->Route->waypoint->position->relativelaneposition (reset the start point and the end point), I solved the problem of the object on the straight road.