carla-simulator/scenario_runner #707: How to add the behaviors to behavior tree.

tecena, 2020-12-15:
Hello,

    I'm facing problem in writing the new python script while creating new scenario. I understood that there 4 parts/methods in each scenario file. I'm facing trouble in adding behaviors to the behavior tree. I'm able to add needed behaviors for the actors. But not getting clear idea on adding behaviors in parallel and sequential to the tree. which all behaviors are to be added in sequence and in parallel to the tree. Please help!

akshatamk, 2020-12-21:
@tecena @even I'm facing the same issue. Not getting clear idea on how to build the behavior tree . I can create the behaviors. But not clear understanding on how to build the behavior tree. Please Help!

GeneralMak, 2021-07-27:
Hello,is your problem solved? I think I have encountered a problem similar to yours, and the creation of a behavior tree is a difficult problem.
When I asked their developers to solve the problem ,they provided a new solution:(Why not using OpenSCENARIO? This is much easier, and there is a graphical scenario editor: https://github.com/carla-simulator/traffic-generation-editor?)

closed 2021-11-23 by glopezdiest