carla-simulator/scenario_runner #710: Autopilot does not follow the route

kuasho, 2020-12-16:
Dear all,

I am debugging the scenarios of the leaderboard using the scenario runner for other purposes than the autonomous driving.

I set the agent in autopilot using the command P to start the navigation, and I realized that it does not follow the route that is set on the .xml, which I can see on the image using the flag --debug as some markers

 - OS: [e.g. Ubuntu 18.04]
 - CARLA Version [e.g. CARLA 0.9.10]
 - Python version [e.g. 3.7]
 - Version [e.g. 0.9.10]

Thanks in advance

glopezdiest, 2020-12-17:
Hey @kuasho, I understand you are using the autopilot function of the manual control? If that's the case, when activating it, the autopilot uses the TrafficManager, which makes the vehicles randomly move around the town. That feature has nothing to do with the leaderboard's route. I'd suggest using the [*npc_agent*](https://github.com/carla-simulator/leaderboard/blob/master/leaderboard/autoagents/npc_agent.py), which will indeed automatically follow the route

kuasho, 2020-12-17:
Thanks for the advice @glopezdiest

Does the npc_aget allow a manual control using the keyboard?
Or I have to mix both codes in order to get these two driving behaviours?

glopezdiest, 2020-12-17:
> Or I have to mix both codes in order to get these two driving behaviours?

Indeed, the npc_agent is completely autonomous and doesn't support keyboard controls. If you want to do that, I'd recommend merging the npc_agent into the human_agent.

kuasho, 2020-12-17:
> > Or I have to mix both codes in order to get these two driving behaviours?
> 
> Indeed, the npc_agent is completely autonomous and doesn't support keyboard controls. If you want to do that, I'd recommend merging the npc_agent into the human_agent.

Thanks for your answer, I really appreciate

closed 2021-02-12 by fabianoboril

Nitro60zeus, 2023-08-24:
> > Or I have to mix both codes in order to get these two driving behaviours?
> 
> Indeed, the npc_agent is completely autonomous and doesn't support keyboard controls. If you want to do that, I'd recommend merging the npc_agent into the human_agent.

Hi !
Thanks for this. I am able to run the route in autonomous mode. However I have two issues:
1> The speed is very less (12kmph)
2> The traffic is behaving weirdly and crashing here and there.

Can you please guide me a bit? It will be extremely helpful !!