carla-simulator/scenario_runner #712: Runtime error: map not found when using open scenario

tangbaoGoGo, 2020-12-22:
Hi，I'm trying to use Scenario Runner to run the OpenScenario scenarios and use Open Drive maps. I created a simple road scene in Road Runner and exported it by CARLA option. I decided to test the scene by ChangingWeather.xosc example in /srunner/examples, so I changed the "RoadNetwork" to below: 

[image]

then run with this command:
D:\CARLA_0.9.10.1\scenario_runner-0.9.10>python scenario_runner.py --openscenario D:\RoadRunner_Trial\Exports\ChangingWeather.xosc --reloadWorld

**The scene is loaded in CARLA, but lost the lane markings. Also the terminal reports this:**

D:\CARLA_0.9.10.1\scenario_runner-0.9.10>python scenario_runner.py --openscenario D:\RoadRunner_Trial\Exports\ChangingWeather.xosc --reloadWorld
_run_openscenario
WARNING:[SR:OpenScenarioConfiguration]: Wrong OpenDRIVE map in use. Forcing reload of CARLA world
WARNING: cannot parse georeference: ''. Using default values.
WARNING: cannot parse georeference: ''. Using default values.
D:\RoadRunner_Trial\Exports\test.xodr
Traceback (most recent call last):
  File "scenario_runner.py", line 601, in <module>
    sys.exit(main())
  File "scenario_runner.py", line 592, in main
    result = scenario_runner.run()
  File "scenario_runner.py", line 497, in run
    result = self._run_openscenario()
  File "scenario_runner.py", line 487, in _run_openscenario
    result = self._load_and_run_scenario(config)
  File "scenario_runner.py", line 349, in _load_and_run_scenario
    if not self._load_and_wait_for_world(config.town, config.ego_vehicles):
  File "scenario_runner.py", line 299, in _load_and_wait_for_world
    self.world = self.client.load_world(town)
RuntimeError: map not found

**Screenshots**
[image]
[image]
[image]

Can anyone see where the problem is? 
Thank you for your help!

**Desktop:**
 - OS: Win10
 - CARLA Version: 0.9.10.1
 - Python version: 3.7.9
 - Scenario_runner Version: 0.9.10

fabianoboril, 2020-12-22:
Hi,
this seems to be issue inside CARLA. Can you maybe share the xodr file? Then we could take a closer look. @glopezdiest Do you have a direct idea what the issue could be?

glopezdiest, 2020-12-22:
In short, ScenarioRunner isn't meant to be used with OpenDRIVE maps, which makes the API use the wrong function.

If you want to load a **CARLA town**, you have to use `client.load_world(town_name)`, which is the way ScenarioRunner does it.
```
self.world = self.client.load_world(carla_map)
```

However, if you want to use an **OpenDRIVE map**, you have to use `client.generate_opendrive_world(odr_file)`.
```
with open(map_path) as odr_file:
	self.world = self.client.generate_opendrive_world(odr_file.read())
```

> The scene is loaded in CARLA, but lost the lane markings.

That's how the OpenDRIVE maps are created, which is the second image on your post, and yes, the lane markings aren't shown.

tangbaoGoGo, 2020-12-24:
> In short, ScenarioRunner isn't meant to be used with OpenDRIVE maps, which makes the API use the wrong function.
>
> If you want to load a **CARLA town**, you have to use [...]

tangbaoGoGo, 2020-12-24:
Hi Thank you for the answers! 

By "ScenarioRunner isn't meant to be used with OpenDRIVE maps", do you mean that I should not reference .xodr file in `<RoadNetwork>` tag, but should reference "Town01" or some other map name? 

I read the "Add new map" section in CARLA documentation and found that I still need to import the .fbx & .xodr & .xml files (from RoadRunner) into Unreal Engine and finally export a map package. So I think the correct way is to have this map package, then I can reference to it in `<RoadNetwork>` just as the examples reference "Town01". Is this right?

glopezdiest, 2021-01-07:
> By "ScenarioRunner isn't meant to be used with OpenDRIVE maps", do you mean that I should not reference .xodr file in <RoadNetwork> tag, but should reference "Town01" or some other map name?

Technically yes, the way it is right now, you can only use CARLA maps, not OpenDRIVE maps. In order to support xodr maps, you should change the start of *_load_and_wait_for_world()* function at *scenario_runner.py*, to something like this:

```
if map_path.endswith('.xodr'):
    with open(map_path) as odr_file:
	self.world = self.client.generate_opendrive_world(odr_file.read())
else:
    self.world = self.client.load_world(town)
```

fabianoboril, 2021-01-07:
> > By "ScenarioRunner isn't meant to be used with OpenDRIVE maps", do you mean that I should not reference .xodr file in  tag, but should reference "Town01" or some other map name?
> 
> Technically yes, the way it is right now, you can only use CARLA maps, not OpenDRIVE maps. In order to support xodr maps, you should change the start of __load_and_wait_for_world()_ function at _scenario_runner.py_, to something like this:
> 
> ```
> if map_path.endswith('.xodr'):
>     with open(map_path) as odr_file:
> 	self.world = self.client.generate_opendrive_world(odr_file.read())
> else:
>     self.world = self.client.load_world(town)
> ```

No. Using OpenSCENARIO, you can use any OpenDRIVE map directly. It is ensured in https://github.com/carla-simulator/scenario_runner/blob/e99f4dede78f3168248d8db0b1fc6b099469dcee/srunner/scenarioconfigs/openscenario_configuration.py#L162  that the map is correctly loaded.

glopezdiest, 2021-01-08:
@fabianoboril I definitely didn't remember that. However, at the openscenario configuration, you are just storing the *town* information, whereas the loading of the map is still always done through the `load_world()`, function, which doesn't accept xodr maps.

glopezdiest, 2021-01-11:
Okay, I could finally test it and indeed it works well with OpenDrive maps.

@tangbaoGoGo However, for this issue specifically, the problem lies in that you are using the `reloadWorld` with OpenSCENARIO, which isn't needed as it automatically loads the correct map. This `reloadWorld` does always use the *load_world* function, which is why is fails when using an xodr. Just remove it and the problem will be solved.

closed 2021-03-17 by fabianoboril