carla-simulator/scenario_runner #717: Throws an error while running manual_control.py

athulg, 2021-01-18:
Hi, I followed all the steps accordingly but when I try running manual_control.py it throws the below error. I also double checked the paths and environment variables.

I have Python 2.7 and CARLA 0.9.10
```
Traceback (most recent call last):
  File "manual_control.py", line 58, in <module>
    from examples.manual_control import (World, HUD, KeyboardControl, CameraManager,
ImportError: No module named examples.manual_control
```
Thank you!

athulg, 2021-01-18:
I solved this, I just had to add an init.py file on the top of ~/PythonAPI/examples folder.

Now when I run scenario_runner.py I get another error

```
  File "/home/scenario_runner/srunner/scenariomanager/scenarioatomics/atomic_behaviors.py", line 2230, in initialise
    self._actor.set_target_velocity(carla.Vector3D(0, 0, 0))
AttributeError: 'Vehicle' object has no attribute 'set_target_velocity'
'Vehicle' object has no attribute 'set_target_velocity'
```

Thanks!

glopezdiest, 2021-01-19:
Hey, CARLA 0.9.9 uses the `carla.Vehicle.set_velocity` function but, when we updated to 0.9.10, it was changed to `carla.Vehicle.set_target_velocity`. We updated SR to also reflect this. From your error, seems like you are using CARLA 0.9.9 (or lower) and SR 0.9.10? Is this the case?

AthulGangadharan, 2021-01-19:
Hello @glopezdiest , Thank you for the response, I've Unreal Engine version 4.24.3, Carla version 0.9.10 & SR 0.9.10

athulg, 2021-01-19:
Hello @glopezdiest, thanks for the response, I've Unreal Engine version 4.24.3, Carla version 0.9.10 & SR 0.9.10

glopezdiest, 2021-01-19:
> But, it works fine with SR 0.9.9

That strongly leads me to believe that your CARLA isn't 0.9.10, as those are incompatible (due to these velocity API changes). Maybe, what you can do in order to make sure you are using 0.9.10 is to go to `carla/PythonAPI/carla/source/libcarla/Actor.cpp`, where you'll find some of the Python API mapping to the C++ functions. The `set_target_velocity` mapping is around line 100. Check whether or not it is mapped to `set_target_velocity` (0.9.10) or `set_velocity` (0.9.9)

athulg, 2021-01-28:
It turns out I had set_velocity (0.9.9) but my CHANGELOG in CARLA directory anyways showed I had 0.9.10. Anyways, I re installed CARLA 0.9.10 and it works fine. Thank you!

closed 2021-01-28 by athulg

SiNa-Dehbari, 2021-03-15:
Hello everybody,

sorry for ew-opening.
I hvave the same issue as @athulg 
I receive this error:

`Traceback (most recent call last):
  File "manual_control.py", line 58, in <module>
    from examples.manual_control import (World, HUD, KeyboardControl, CameraManager,
ImportError: No module named examples.manual_control`

recently, I have upgraded to Carla 9.11 and SR 9.11

could you please let me know what should I do?

glopezdiest, 2021-03-16:
Hey @SiNa-Dehbari, recent version need an additional PYTHONPATH link:

`export PYTHONPATH=$PYTHONPATH:$CARLA_ROOT/PythonAPI`

which seems to be the one you are missing

Jia-Wei-Wang, 2021-09-22:
I add  this line to ~/.bashrc

`export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/examples`

then it works.

whitehatjunior35, 2023-01-23:
Hi, I have been facing this issue a lot lately can someone help me with this? 

Traceback (most recent call last): File "[manual_control.py](https://manual_control.py/)", line 1137, in <module> main() File "[manual_control.py](https://manual_control.py/)", line 1129, in main game_loop(args) File "[manual_control.py](https://manual_control.py/)", line 1045, in game_loop world = World(client.get_world(), hud, args) RuntimeError: time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to [127.0.0.1:2000](https://127.0.0.1:2000/)