carla-simulator/scenario_runner #718: LaneChange manoeuvre unstable

jasonxzzhang, 2021-01-20:
Hi, first of all thanks for all the effort input into make the scenario runner, I personally find it is very useful for my research. I would like to report an issue I have while running a scenario using python file.  I have pasted the sequence for one of the actors in my scenario. From the second lane change, it starts losing control and oscillating with an increasing amplitude, which causes not able to complete the third lane change.  

**Expected behavior**
The expected behaviour will be DriveForward (using a timer of 5s) --> LaneChange left (for 1 lane) --> DrivieForward (using a timer for 5s) --> LaneChange right (for 1 lane) --> LaneChange right (for 1 lane) --> DriveForward (for 5s) --> LaneChange left --> DriveForward. However after the second lane change it lost control. I suspect it might be due to the previous waypoint follower not terminated? i have also attached a py_trees debuging image.

**Screenshots, code**
[image]

`sequence_1 = py_trees.composites.Sequence("1")

        sequence_1.add_child(ActorTransformSetter(self.other_actors[0], self._transform_visible1))
        just_drive = py_trees.composites.Parallel("DrivingForward", policy=py_trees.common.ParallelPolicy.SUCCESS_ON_ONE)
        v1_drive_forward = WaypointFollower(self.other_actors[0], 20)
        just_drive.add_child(v1_drive_forward)
        just_drive.add_child(LocalTimer(self.other_actors[0], 5))
        sequence_1.add_child(just_drive)
        sequence_1.add_child(LaneChange(self.other_actors[0], direction='left', speed=20.11675, distance_other_lane=40.2335))
        just_drive = py_trees.composites.Parallel("DrivingForward", policy=py_trees.common.ParallelPolicy.SUCCESS_ON_ONE)
        v1_drive_forward = WaypointFollower(self.other_actors[0], 24)
        just_drive.add_child(v1_drive_forward)
        just_drive.add_child(LocalTimer(self.other_actors[0], 5))
        sequence_1.add_child(just_drive)
        sequence_1.add_child(LaneChange(self.other_actors[0], direction='right', speed=20, distance_other_lane=40))
        sequence_1.add_child(LaneChange(self.other_actors[0], direction='right', speed=20, distance_other_lane=40))
        just_drive = py_trees.composites.Parallel("DrivingForward", policy=py_trees.common.ParallelPolicy.SUCCESS_ON_ONE)
        v1_drive_forward = WaypointFollower(self.other_actors[0], 20)
        just_drive.add_child(v1_drive_forward)
        just_drive.add_child(LocalTimer(self.other_actors[0], 5))
        sequence_1.add_child(just_drive)
        sequence_1.add_child(LaneChange(self.other_actors[0], direction='left', speed=20, distance_other_lane=40))
        v1_drive_forward = WaypointFollower(self.other_actors[0], 20)
        sequence_1.add_child( v1_drive_forward)`

**Desktop (please complete the following information):**
 - OS: [Ubuntu 20.04]
 - CARLA Version [0.9.10.1]
 - Python version [3.7]

jasonxzzhang, 2021-01-21:
I think this might due to the synchronisation issue, I tried running scenario_runner in --sync mode, it gets much better however agent will still oscillate after changing lane but eventually it becomes stable

fabianoboril, 2021-03-17:
Depending on the speed this oscillation is normal.

glopezdiest, 2021-11-23:
As @fabianoboril said, this might have been due the controllers, causing lane changes to not be done smoothly. We have been updating them quite a bit recently, so is this still happening? I'm preemptively closing this issue as this is quite an old post, but feel free to reopen it if needed