carla-simulator/scenario_runner #719: ScenarioRunner with Carla v0.9.11

semmelknoedel, 2021-01-26:
Hello,
is there a version that's compatible with Carla version 0.9.11?
I updated Carla to 0.9.11 but did not think that there might be no scenario_runner version, yet. The master is not working for me.
Carla alone is running without a problem, though.

glopezdiest, 2021-01-27:
Hey @semmelknoedel, the tag is not there but we did the necessary changes to support 0.9.11. What issues are you seeing with current master and CARLA 0.9.11?

semmelknoedel, 2021-01-27:
Hi @glopezdiest, i get the following terminal output:

python3 scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld
Configuration for scenario FollowLeadingVehicle_1 cannot be found!
No more scenarios .... Exiting

But after rolling back to version 0.9.10.1 I have the same problem. So I will check if there is something else wrong.

glopezdiest, 2021-01-27:
That might be caused due to the lack of some environmental variables. Can you make sure that you have a *SCENARIO_RUNNER_ROOT* pointing to your scenario runner folder?

semmelknoedel, 2021-01-27:
Your were right. I lately changed something in my Paths and then the SCENARIO_RUNNER_ROOT was wrong. Thanks its working now!

glopezdiest, 2021-01-27:
Perfect! Closing this issue. However, feel free to open it again if the problem persists

closed 2021-01-27 by glopezdiest

SExpert12, 2023-06-20:
hi everyone,

I got the same error:
python3 scenario_runner.py --scenario follow_leading_vehicle.py --reloadWorld

  warnings.warn(msg, UserWarning)
Configuration for scenario follow_leading_vehicle.py cannot be found!
No more scenarios .... Exiting

I set my path variables in .bashrc file as:
export CARLA_ROOT=/home/sona/Downloads/CARLA14
export SCENARIO_RUNNER_ROOT=/home/sona/Downloads/CARLA14/PythonAPI/carla/scenario_runner
export PYTHONPATH=/usr/bin/python3:/home/sona/Downloads/CARLA14/PythonAPI/carla/dist/carla-0.9.14.-py3.7-linux-x86_64.egg
export PYTHONPATH=/usr/bin/python3:/home/sona/Downloads/CARLA14/PythonAPI/carla/agents
export PYTHONPATH=/usr/bin/python3:/home/sona/Downloads/CARLA14/PythonAPI/carla
export PYTHONPATH=/usr/bin/python3:/home/sona/Downloads/CARLA14/PythonAPI

any suggestion to solve this error