carla-simulator/scenario_runner #720: OpenScenario <SpeedAction> Attributes  support

Vela-zz, 2021-01-27:
When I using Scenario Runner run a  openscearnio event,  like 

```
 <LongitudinalAction>
    <SpeedAction>
       <speedActionDynamics dynamicsShape='linear' dynamicsDimension='rate' value='5'>
        ......

```
which means the Target vehicle should accelerate or decelerate at 5m/s^2  untill target speed,
But in carla, especially on curve road,  when run by scenario runner, the IMU sensor reflects that the true decelerate speed is  only 2m/s^2 and not reach 5m/s^2, can scenario runner with carla control it precisely ?

docs in https://carla-scenariorunner.readthedocs.io/en/latest/openscenario_support/  shows, the Speed Action tag is all green check without any notes

I can only find some content about scenario runner realize SpeedAction Tag through behiavior tree, by using  `class ChangeActorTargetSpeed(AtomicBehavior) `, but how carla check behiavior tree and controll the vehicle ?

I found ` class AccelerateToVelocityand(AtomicBehavior)`  control vehicle by set actor.throttle, but I can not find any reference of this class in openscenario parser part

thanks ...

glopezdiest, 2021-01-27:
Hey @Vela-zz I think this is gonna be a long post so let's break it down in sections.

> docs in https://carla-scenariorunner.readthedocs.io/en/latest/openscenario_support/ shows, the Speed Action tag is all green check without any notes

Most of the OSC actions dynamic transitions are not supported. This is definitely reflected in the docs of the more recent actions, but it might be missing for some of the first one (such as Speed Action). This is not supported because all these actions are generally achieved by using controllers, which aren't complex enough to force these specific behaviors. 

> I can only find some content about scenario runner realize SpeedAction Tag through behiavior tree, by using class ChangeActorTargetSpeed(AtomicBehavior) , but how carla check behiavior tree and controll the vehicle ?

As I just said, everything is done via controllers and what *ChangeActorTargetSpeed* does is overwrite the target speed of the controller with the desired one. As you can see, nothing is being passed regarding the dynamic shape

> I found class AccelerateToVelocityand(AtomicBehavior) control vehicle by set actor.throttle, but I can not find any reference of this class in openscenario parser part

This is an old behavior that we used, but one of its main drawbacks is that it overwrittes any control already calculated, or in other words, it can't accelerate and turn at the same time.

> But in carla, especially on curve road, when run by scenario runner, the IMU sensor reflects that the true decelerate speed is only 2m/s^2 and not reach 5m/s^2, can scenario runner with carla control it precisely ?

So, to sum up, this isn't yet supported. If you want to support it, what you might to do is something like this:
1) parse the OSC values at the *openscenario_parser.py* and pass them to `ChangeActorTargetSpeed`
2) at the `ChangeActorTargetSpeed.update`, keep calculating the desired speed at each frame, according to your dynamic shape
3) update the target speed of the actor controller `actor_dict[self._actor.id].update_target_speed(desired_value)`

Hope this helps :slightly_smiling_face:

Vela-zz, 2021-01-27:
> Hey @Vela-zz I think this is gonna be a long post so let's break it down in sections.
>
>
> Most of the OSC actions dynamic transitions are not supported. This is definitely reflected in the docs of the [...]

Thanks, I will try it

closed 2021-01-27 by Vela-zz

TigerStone93, 2022-10-27:
I think this problem should be stated clearly in the documentation.

In the link below, there is no caution on the notes of table.
[https://carla-scenariorunner.readthedocs.io/en/latest/openscenario_support/](url)

Since there is no mention of not supporting the `dynamicsShape` and `dynamicsDimension` on the `SpeedAction`, I thought it would be support without doubt.