carla-simulator/scenario_runner #722: RuntimeError: trying to create rpc server for traffic manager; but the system failed to create because of bind error.

athulg, 2021-01-28:
Hello, I'm facing this error now every time I run scenario_runner.py, I installed SR again and tried scenario_runner.py and then it worked fine only once and now I again get this error.

```
Traceback (most recent call last):
  File "scenario_runner.py", line 600, in <module>
    sys.exit(main())
  File "scenario_runner.py", line 589, in main
    scenario_runner = ScenarioRunner(arguments)
  File "scenario_runner.py", line 91, in __init__
    self.traffic_manager = self.client.get_trafficmanager(int(self._args.trafficManagerPort))
RuntimeError: trying to create rpc server for traffic manager; but the system failed to create because of bind error.
```

Thank you!

 - OS: [Ubuntu 18.04]
 - CARLA Version CARLA 0.9.10]
 - Python version [3.6]
 - Version [e.g. 0.9.10]

sagar-g-v, 2021-02-02:
@athulg which scenario_runner version are you using?

if You are using carla 0.9.10 use the scenario runner version 0.9.10

[Version 0.9.10](https://github.com/carla-simulator/scenario_runner/releases/tag/v0.9.10) and the 0.9.10 Branch: Compatible with [CARLA 0.9.10](https://github.com/carla-simulator/carla/releases/tag/0.9.10)

athulg, 2021-02-02:
@sagar-g-v I'm using 0.9.10 CARLA with 0.9.10 SR

glopezdiest, 2021-02-11:
Hey @athulg, sorry for the delay. that isn't related to SR, but CARLA instead. Basically, you are trying to create the TrafficManager in a port that is busy. This tends to be caused by a  previous run of SR that hasn't been properly terminated, which means that the TM is still active. How are you running scenario runner exactly? Could you try using the `--trafficManagerPort` argument to another number?

Esquilli, 2021-02-11:
This issue happens to me sometimes. I just run `pkill -9 python` and then run everything again.

closed 2021-11-23 by glopezdiest