carla-simulator/scenario_runner #724: manual_control.py example

Fady-Maged-Louis, 2021-01-31:
**Describe our question or idea**
I run manual_control.py example.

**Expected behavior**
pygame window flashes then close and error appear

```
 File "manual_control.py", line 1187, in <module>
    main()
  File "manual_control.py", line 1179, in main
    game_loop(args)
  File "manual_control.py", line 1095, in game_loop
    world = World(client.get_world(), hud, args)
RuntimeError: time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:10000
```

**Desktop 
 - OS: windows10
 - CARLA Version :CARLA 0.9.10
 - Python version 3.7.9
[image]

sagar-g-v, 2021-02-02:
@Fady-Maged-Louis Try giving 10-20 sec timeout 
search for ` client.set_timeout(2.0)` in manual_control and change it to higher value

Fady-Maged-Louis, 2021-02-03:
unfortunately, the same problem

glopezdiest, 2021-02-11:
Hey @Fady-Maged-Louis, could you provide some more information?

How are you running the server? Your error is related to the client not being able to connect to the server.

Are you using [ScenarioRunner's manual control](https://github.com/carla-simulator/scenario_runner/blob/master/manual_control.py) or [CARLA's one](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/manual_control.py)?

Fady-Maged-Louis, 2021-02-11:
I use CARLA's one 
I try to run some examples like spawn_npc.py and dynamic_weather.py they run perfectly but manual_control.py don't run

glopezdiest, 2021-02-11:
I'll take a look then. Not sure what could be the issue right now, as they both have the same method of getting the CARLA egg and creating the client.

PD: Also, transferring this issue to CARLA as this isn't ScenarioRunner related