carla-simulator/scenario_runner #726: Inconsistency in speed for CARLA Blueprint for motorbike "vehicle.yamaha.yzf"

Sumza, 2021-02-09:
Inconsistency in speed(constant fluctuation of speed) is observed for the CARLA Blueprint "vehicle.yamaha.yzf" which results in uneven movement of vehicle.

In the below video file, we can see that the motorbike speed fluctuates between the range 50 - 60 Kmph while the speed is set to 60Kmph in the xosc file.

https://user-images.githubusercontent.com/68461250/107344800-e2811780-6ae8-11eb-848a-71330f920bbc.mp4

Is there any updation in-progress for the vehicle dynamics to avoid the inconsistency in speed ?

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - CARLA Version [e.g. CARLA 0.9.4]
 - Python version [e.g. 2.7]
 - Version [e.g. 0.9.4]

fabianoboril, 2021-02-12:
Hi,
I guess you are using the default controller for the vehicle, or did you specify a dedicated OSCController? If you use the default one, this may explain the observation. Would it be possible to use the simple_vehicle_controller? You can assign a controller like shown here:
https://github.com/carla-simulator/scenario_runner/blob/e735586789ea28b4a76ce8cb6a54fa3c28a7d7f7/srunner/examples/OscControllerExample.xosc#L101

Sumza, 2021-02-24:
Hi, I tried assigning the same controller as mentioned by you, 
but i observed the same behavior again.

Can you guide me if it can be solved in any other way?

fabianoboril, 2021-03-17:
Maybe a stupid question, but is the manual_control attached to the bike? If yes, did you remove the actuator controlling code from manual_control?

glopezdiest, 2021-11-23:
Closing this issue as both controllers and physics have been improved quite a bit in order to smooth out these behaviors. Feel free to reopen it if this is still happening to you

closed 2021-11-23 by glopezdiest