carla-simulator/scenario_runner #727: Vehicle Maneuver Stops After LaneChange Event

sagar-g-v, 2021-02-09:
Hi,
I am trying to run OSC scenario with Lateral and Longitudinal events in parallel but I am not getting the expected results

below are my test scenario

1) accelerate --> lane chage --> decelerate
2) accelerate --> lane change --> lane change --> decelerate
3) accelerate --> lane change --> lane change --> accelerate --> decelerate

In all the cases vehicle does not perform the event that comes after lane change

Here is my scenario file 
[LateralLongitudinalExample.txt](https://github.com/carla-simulator/scenario_runner/files/5953693/LateralLongitudinalExample.txt)

please check the sceanrio and kindly let me know if i have missed anything.

Regards,
Sagar

Lu7913, 2021-02-16:
I'm experiencing similar issue #654. Also some problems with lane change: #718

sagar-g-v, 2021-03-02:
Hi,
I found the cause of this issue

https://github.com/carla-simulator/carla/blob/master/PythonAPI/carla/agents/navigation/local_planner.py

After exausting `self._waypoints_queue` reseting `self._global_plan` to `False` is not done.
```python
if len(self._waypoints_queue) == 0 and len(self._waypoint_buffer) == 0:
    control = carla.VehicleControl()
    control.steer = 0.0
    control.throttle = 0.0
    control.brake = 1.0
    control.hand_brake = False
    control.manual_gear_shift = False
    return control
```

so when I added `self._global_plan = False` the issue is resolved.

```python
if len(self._waypoints_queue) == 0 and len(self._waypoint_buffer) == 0:
    self._global_plan = False
    control = carla.VehicleControl()
    control.steer = 0.0
    control.throttle = 0.0
    control.brake = 1.0
    control.hand_brake = False
    control.manual_gear_shift = False
    return control
```

glopezdiest, 2021-03-04:
Hey @sagar-g-v. Sorry for the huge response delay but yes that is indeed the case. _Warning: Long explanation incoming_.

By default, the local planner works by randomly moving the actor through town, while automatically populating the waypoint buffer so that it never ends. Additionally, a specific route can be given by the user (see [this function](https://github.com/carla-simulator/carla/blob/master/PythonAPI/carla/agents/navigation/local_planner.py#L197)). However, this last behavior makes the local planner stop populating the waypoint buffer (last line of the previous link causes [this conditional](https://github.com/carla-simulator/carla/blob/master/PythonAPI/carla/agents/navigation/local_planner.py#L233) to fail) and therefore, once the given route ends, the vehicle stops. This is indeed what you've reported at the previous comment.

OSC lane changes do exactly this, they set the global plan to the specific trajectory of the lane change. By default, these lane changes last for 1000 meters (check [here](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/tools/openscenario_parser.py#L1042)) after the lane change is complete, so this isn't an issue in most cases, except when overwritten by another lateral action (such as lane offset).

Right now, the best solution would be to comment [this line](https://github.com/carla-simulator/carla/blob/master/PythonAPI/carla/agents/navigation/local_planner.py#L221), which will make the local planner never stop populating the waypoint buffer, even if a specific route is given. This is similar to your approach, but it avoids having to reinitialize everything. 

> I could create a PR but it is on carla repo. (it may take more time to review)

If you can, I'd like you to try this approach and see if it fixes your issues. I'll keep testing this and see if it is a good idea to optionally stop the waypoint buffer populating, instead of being always done. If so, I'll do a PR to the main CARLA repo. If not, we could just do a patch at SR as this indeed breaks the LaneChange behavior.

sagar-g-v, 2021-03-05:
Thank you for the detailed explaination. I will check and update my status here

fabianoboril, 2021-03-17:
@sagar-g-v @glopezdiest There might be another option. We could repopulate the waypoint list from SR side, when we detect that the lane change is completed.

PR #741 (2021-03-17): Fixed never ending lane change maneuver in OSC
Fixes #727 

Past behavior:
A lane change in OSC was always set to cover 1000m, if there was no proper end condition set.

New behavior:
A lane change in OSC covers now only 10m on the new lane. To avoid that a vehicle stops immediately after, the behavior returns with success but provides waypoints for another 200 meters to allow the vehicle to continue moving.

#### Where has this been tested?

  * **Platform(s):** Ubuntu 20.04
  * **Python version(s):** 3.8
  * **CARLA version:** 0.9.11

---
This change is [[image]](https://reviewable.io/review [... cut]
closed 2021-03-17 by fabianoboril