carla-simulator/scenario_runner #730: Metric module with Docker Container

semmelknoedel, 2021-02-22:
Hello,

when I use the metrics tool of the scenario runner (on my local machine) and running the Carla server in a docker container (remote machine), there is no .log file created. Is there a way to solve that problem? Is it because the Carla server runs on a remote system or because it runs in a docker container?

 - OS: Ubuntu 18.04
 - CARLA 0.9.11
 - Python version 3.6

glopezdiest, 2021-02-24:
Hey @semmelknoedel. this issue might not be related to SR, but the CARLA recorder itself. Could you please try it using the [CARLA recorder example script](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/start_recording.py), and see if the *.log* file is being generated? Also, the log file might be erased once the docker is closed or you might not have the necessary permissions, could you check that that isn't the case?

semmelknoedel, 2021-03-03:
@glopezdiest  I tried it now with the recording tool. It says a file is created on my server machine, but the given folder is empty. I did not close the docker container, so this should not be the issue. I also ran the script on the server machine. The file is not generated there as well. However, there is one created inside the docker container. Is this the same file the metric tool is using? 
Maybe I was a bit confused, since the user on my remote machine is also called carla, same as the one inside the docker container.

MisterioRemo, 2021-05-11:
@semmelknoedel hello.
I had a similar problem. But I ran the client and server on a local machine.
I looked in the `scenario_runner.py` to know where the file should be created: 
```py
if self._args.record:
   recorder_name = "{}/{}/{}.log".format(
      os.getenv('SCENARIO_RUNNER_ROOT', "./"), self._args.record, config.name)
   self.client.start_recorder(recorder_name, True)
```
And noticed that I didn't set a variable `SCENARIO_RUNNER_ROOT`, so my recorder_name looked like `.//Logs/FollowLeadVehicle_1.log`. After I set `SCENARIO_RUNNER_ROOT`, log file were created.

semmelknoedel, 2021-05-11:
It is not working over network. I didn't change anything except first running a local carla instance. There it worked fine. After that I started the docker container on my remote machine with (port forwarding) and there is no .log-file.

glopezdiest, 2021-05-11:
@semmelknoedel How are you running the server? I do remember that a while back I tried running a docker with the server outside of it and, as the recorder is being created by the server, all log files were also created outside of the docker. Can this be your issue?

semmelknoedel, 2021-05-11:
@glopezdiest loacally, I run the release version (0.9.11). I don't have an nvidia card so docker is not possible. On my remote machine I start the released docker docker container (0.9.11). With both setups I also start the scenario_runner.

What exactly do you mean by docker container with the server outside of it?

glopezdiest, 2021-05-12:
Okay, I just reread your comment and I basically completely misunderstood your issue. I meant the opposite of what you are doing (Running SR inside a docker and the CARLA server on my local machine). 

After my first comment, you run the recorder without SR and you got a file inside the docker
> I also ran the script on the server machine. The file is not generated there as well. However, there is one created inside the docker container

That IS indeed the *.log* file recorded by the server. As the recorder is created by the server, and your server is inside a docker, it will always be created inside the docker, regardless of where your client is (and the recorder doesn't complain if the path is wrong, so...). For your case, as @MisterioRemo said, SR sets the path as `recorder_name = "{}/{}/{}.log".format(os.getenv('SCENARIO_RUNNER_ROOT', "./"), self._args.record, config.name)`, so you have to make sure that that path is correct inside the docker, not at your local machine.

To solve it, make sure that you have *SCENARIO_RUNNER_ROOT* defined inside your docker, or remove it and just don't use that variable altogether (we didn't think about your use-case when setting that path and it's pretty much useless for you, sorry about that)

glopezdiest, 2021-11-23:
Solved, closing the issue. Feel free to reopen if if that isn't the case