carla-simulator/scenario_runner #731: Ego vehicle not found in manual_control

DeepAnonymous, 2021-02-26:
Hi all,

I am new to scenario_runner, and I just want to run manual_control.py

However, the program just keep looping around "Waiting for the ego vehicle..."

`possible_vehicles = self.world.get_actors().filter('vehicle.*')
`
I tried to print `len(possible_vehicles)` and it shows **0**.

That means it cannot find any vehicle with the name `hero`.

Please help.

Thank you very much.

glopezdiest, 2021-03-03:
Hey @DeepAnonymous, you have to first run a scenario with `python scenario_runner.py --scenario <scenario_name>` and then run that manual control on another terminal, to connect to the scenario you just started. Take into account that scenarios tend to have a 60s timeout, and after that, they'll be automatically stopped.

DeepAnonymous, 2021-03-07:
Thank you for your reply.

closed 2021-11-23 by glopezdiest