carla-simulator/scenario_runner #732: Why ActorDestroy and `remove_all_actors()` are both called in scenario classes?

youngsend, 2021-03-01:
Hi, Carla team!
I find that in most scenario classes, `ActorDestroy` method is used in the end of the scenario behavior sequence (`_create_behavior`).
I wonder if this is necessary because `self.remove_all_actors()` is always called in `__del__()`.

**Desktop (please complete the following information):**
 - OS: [Ubuntu 16.04]
 - CARLA Version [CARLA 0.9.11]
 - Python version [3.5]
 - Version [0.9.11]

glopezdiest, 2021-03-03:
Hey @youngsend, we general methodology is to use the `ActorDestroy` for actors that end their behavior in weird places (such as bikes in the middle of the road). On the other side, the fact is that trying to remove an actor that is already dead doesn't create any issues (such a warning printout), so we also tend to add more `ActorDestroy` than necessary.

youngsend, 2021-03-03:
@glopezdiest, thank you for the explanation! I see.

closed 2021-03-03 by youngsend