carla-simulator/scenario_runner #733: `ChangeAutoPilot` to ego vehicle

youngsend, 2021-03-01:
Hello Carla Team,
It's very convenient to set autopilot parameters for other vehicles by `ChangeAutoPilot`.
Can I use it to ego vehicle too?

I used `ChangeAutoPilot` to change `max_speed` and `auto_lane_change` parameters for `ego_vehicles[0]` (in `_create_behavior()`), and after I do `python manual_control.py -a`, the speed of ego vehicle seems not to change.

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 16.04]
 - CARLA Version [e.g. CARLA 0.9.11]
 - Python version [e.g. 3.5]
 - Version [e.g. 0.9.11]

youngsend, 2021-03-01:
Sorry I made a mistake. `ChangeAutoPilot` works to `ego_vehicles[0]`.

closed 2021-03-01 by youngsend

Georgezxh, 2022-05-07:
> Sorry I made a mistake. `ChangeAutoPilot` works to `ego_vehicles[0]`.

Hi, I wonder that how can you apply the autopilot to ego vehicle? I run the manual_control, then enter P to change to autopilot mode, but it doesn't work,  the ego vehicle doesn't move. I'm crying now.