carla-simulator/scenario_runner #735: AbsoluteTargetLaneOffset not working

sagar-g-v, 2021-03-02:
Hi SR Team,

Thank you for adding support for OSC LaneOffsetAction.

I have tried creating scenarios with this tag but `AbsoluteTargetLaneOffset` is not working as expexted.
vehicle achieves the given offset and returns to default offset `0.0`. but it is supose to continue with the given offset.

```xml
<LaneOffsetAction continuous="true">
    <LaneOffsetActionDynamics dynamicsShape="linear" maxLateralAcc="1" />
    <LaneOffsetTarget>
        <AbsoluteTargetLaneOffset value="1.2"/>
    </LaneOffsetTarget>
</LaneOffsetAction>
```

**Desktop :**
 - OS: Ubuntu 20.04
 - CARLA Version : CARLA 0.9.11
 - Python version :3.8
 - SR: master

glopezdiest, 2021-03-03:
Hey @sagar-g-v, Is your code based on the example at PR #702 (Added support for LaneOffsetAction)? I just tried that XOSC scenario and its variations (absolute and relative positions), and it seems to be working fine by me. If not, could you send me your XOSC file? Also, note that if the offset is high enough to make the vehicle exit its lane, the action will most likely break

sagar-g-v, 2021-03-03:
Hi @glopezdiest I have made some changes to example at PR #702 to show my use case. 
[LaneOffsetActionTest.zip](https://github.com/carla-simulator/scenario_runner/files/6078451/LaneOffsetActionTest.zip)
In this case obsolute lane offset is used immediately after lane change

Sorry i forgot to mention that I have modified SR and local_planner little bit before running this example but it was not working without the modification due to issue #727 

also i made sure lanechange ends at 25 meters after entering target lane by changing `distance_other_lane=25`
https://github.com/carla-simulator/scenario_runner/blob/e735586789ea28b4a76ce8cb6a54fa3c28a7d7f7/srunner/tools/openscenario_parser.py#L1042

glopezdiest, 2021-03-04:
Okay, I'll take a look to both this problem and #727  and report back

glopezdiest, 2021-03-04:
After some tests, I am indeed seeing your problem but it isn't related to the `LaneOffset` but rather a mix between the controller and the `LaneChange`. As such, I'll explain in depth what I found in issue #727.

closed 2021-06-16 by sagar-g-v