carla-simulator/scenario_runner #740: RuntimeError: opendrive could not be correctly parsed (openscenario)

Leviathan321, 2021-03-12:
When I try to run a self defined openScenario file with a self defined road network I get the exception below.
I have put the .xosc and .xodr file together into srunner/examples. The .xodr file is correct, the scenario has been tested with another simulator.

$ python scenario_runner.py --openscenario ./srunner/examples/roundabout-for-carla.xosc 
Traceback (most recent call last):
  File "scenario_runner.py", line 600, in <module>
    sys.exit(main())
  File "scenario_runner.py", line 590, in main
    result = scenario_runner.run()
  File "scenario_runner.py", line 494, in run
    result = self._run_openscenario()
  File "scenario_runner.py", line 482, in _run_openscenario
    config = OpenScenarioConfiguration(self._args.openscenario, self.client)
  File "C:\Program Files\scenario_runner-master\srunner\scenarioconfigs\openscenario_configuration.py", line 59, in __init__
    self._parse_openscenario_configuration()
  File "C:\Program Files\scenario_runner-master\srunner\scenarioconfigs\openscenario_configuration.py", line 90, in _parse_openscenario_configuration
    self._set_carla_town()
  File "C:\Program Files\scenario_runner-master\srunner\scenarioconfigs\openscenario_configuration.py", line 175, in _set_carla_town
    self.client.generate_opendrive_world(str(data))
RuntimeError: opendrive could not be correctly parsed

**Deployment**
 - OS: win10
 - CARLA Version CARLA 0.9.11
 - Python version 3.7
 - Version 0.9.11

fabianoboril, 2021-03-12:
This seems to be related to CARLA parsing the OpenDRIVE file. Can you share this file? Were you able to run CARLA with that OpenDRIVE map?

Best regards,
Fabian

Leviathan321, 2021-03-23:
Hi Fabian,
sy for the late replay, you were right the carla simulator cannot parse it, but the file is correct.
I am waiting that the bug is resolved in carla.
Thanks,

closed 2021-03-25 by Leviathan321