carla-simulator/scenario_runner #748: How to activate the scenario and make ego vehicle act according to the python script like other vehicles (not manual_control)

EchoYJJ, 2021-04-12:
Hi, everyone.

When running scenarios which require choreography of ego vehicle (i.e. in VehicleTurningRight, the route and speed of ego vehicle are arranged), the python manual_control.py couldn't result in expected situation. Even if I run python manual_control.py -a to set the autopilot mode. It won't act according to the python script.(like the video below, ego vehicle should turn right but it stops)
I also have tried to add **scenario_sequence.add_child(ChangeAutoPilot(self.ego_vehicles[0], True))** to my script, but nothing happens, the ego vehicle won't move.

I wonder how to motivate the scenario according to the choreography in python script.
Looking forwards to your reply.
Regards
https://user-images.githubusercontent.com/75726020/114347769-11bf0e00-9b98-11eb-90e3-227a1ef533ea.mp4

**Desktop**
 - OS: Windows
 - CARLA Version 0.9.10
 - Python version 3.7
 - Version 0.9.10

fabianoboril, 2021-04-15:
You can simply use the ego vehicle like any other vehicle inside the python script and assign actions. However, be aware that you have to remove the control-commands from manual_control.py in this case. Otherwise you have two actions being sent to the ego vehicle.

EchoYJJ, 2021-04-15:
Hi @fabianoboril , I've tried this before. I remove Line 148,153,154 which have connection with controller. https://github.com/carla-simulator/scenario_runner/blob/master/manual_control.py#L148
Also, I initialize ego vehicle and create behavior in the example of following_leading_vehicle. As a simplify, I only keep: 1.ActorTransformSetter 2.WaypointFollower 3.ActorDestroy in python script.
[follow_leading_vehicle_ego.txt](https://github.com/carla-simulator/scenario_runner/files/6316853/follow_leading_vehicle_ego.txt)
However, when I run _python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld_ ,the ego vehicle won't move, it loses control from manual_control.py but couldn't act according to python script.

I think maybe the initialization have some problems, do you know how to deal with that?

werewolfdev, 2021-09-03:
Hey @EchoYJJ I have the same issue . have you found a solution for this issue . thanks in advance.