carla-simulator/scenario_runner #749: Error: no module named 'carla'

LucasFKobernic, 2021-04-12:
Hi everyone,

Im trying to test scenario runner, but by trying to run it, i get the error: "no modulenamed carla"

Im using CARLA from package (9.11), so i make the download from scenario runner for the version 9.11 ( Release CARLA Scenario Runner 0.9.11 · carla-simulator/scenario_runner (github.com))

I extracted the file inside the PythonAPI folder.

But by trying the basic example, after running CarlaUE4.exe:

python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld

I got this error:

Traceback (most recent call last):
File “scenario_runner.py”, line 31, in
import carla
ModuleNotFoundError: No module named ‘carla’

The interesting thing is that running any example inside the examples folder, like spawn_npc.py, it works. Can anyone help me with this ?

Thanks!!

**Desktop:**
 - OS: [Windows 10]
 - CARLA Version [CARLA 0.9.11]
 - Python version [3.7.4]
 - Version [0.9.11]

closed 2021-04-13 by LucasFKobernic

joluko, 2021-09-30:
I hi with me the same error as described above appeared. Have you found a solution?

LucasFKobernic, 2021-09-30:
Hi!

I'm not sure anymore, but i think, i forgot to set the variables. In the terminal you should done something like this (changing the path to CARLA root directory and to ScenarioRunner root directory that matches to your case):

set CARLA_ROOT=C:\carla
set SCENARIO_RUNNER_ROOT=C:\carla\scenario_runner-master
set PYTHONPATH=%CARLA_ROOT%\PythonAPI\carla\dist\carla-0.9.11-py3.8-win-amd64.egg
set PYTHONPATH=%PYTHONPATH%;%CARLA_ROOT%\PythonAPI\carla\agents
set PYTHONPATH=%PYTHONPATH%;%CARLA_ROOT%\PythonAPI\carla
set PYTHONPATH=%PYTHONPATH%;%CARLA_ROOT%\PythonAPI

after that scenarioRunner should be able to find CARLA.