carla-simulator/scenario_runner #753: Missing condition in ChangeActorLaneOffset (and other atomics)

semmelknoedel, 2021-04-15:
Since there is a NoneType actor in the joint_actor_list in the openscenario_parser it seems it is necessary to add a additional condition in ChangeActorLaneOffset.termine(). Original code:
`if actor_dict and self._actor.id in actor_dict:`
needs:
`if actor_dict and self._actor is not None and self._actor.id in actor_dict:`

The same seems true for InTriggerDistanceToOSCPosition.update():
Original:
`if osc_transform is not None:`
to
`if osc_transform is not None and self._actor is not None:`

I am not 100% sure if this None actor is the source of the problem, but it seems to me it might cause the errors i get:

AttributeError: 'NoneType' object has no attribute 'id'
'NoneType' object has no attribute 'id'
Resetting ego-vehicle!

I think this is also a problem in other atomic behaviors/trigger that use self._actor

Could this also come from something I have to keep in mind when I build an OpenScenario file? The file is validated an does not seem to go againts OSC standards, but maybe there is something which is not allowed in Carla?

Thanks for your work!

**Desktop:**
 - OS: [Ubuntu 18.04]
 - CARLA Version [CARLA 0.9.11]
 - Python version [3.6]
 - Version [0.9.11]

fabianoboril, 2021-04-20:
Hi,

the part for the maneuver is valid and should be fixed with #755. For the condition, we parse only valid actors. If there is a nontype actor, this indicates another problem. Could you test #755 and see if it solved the issue? If not, please re-open the issue.

semmelknoedel, 2021-04-20:
For some reasons it seems that no actors are given to the Behaviors/Triggers after parsing from the OpenSCENARIO.
However, from the debugger I know all actors are in the joint_actor_list. First I thought it might have to do with the line 255 in open_scenario.py:
`joint_actor_list = self.other_actors + self.ego_vehicles + [None]`

I will have a look with the debugger again an try to find a reason for that.

semmelknoedel, 2021-04-20:
> Hi,
> 
> the part for the maneuver is valid and should be fixed with #755. For the condition, we parse only valid actors. If there is a nontype actor, this indicates another problem. Could you test #755 and see if it solved the issue? If not, please re-open the issue.

Actually, I can't find an option for reopening the issue. But I guess the reason why it is not working might be something else anyway.

fabianoboril, 2021-04-20:
Hi,
check https://github.com/carla-simulator/scenario_runner/blob/9b244deab720e6e2e150f1f5b300cfecf352ba0b/srunner/scenariomanager/carla_data_provider.py#L500 if there is any error returned.

semmelknoedel, 2021-04-20:
I just found out that there is some problem with our used ParameterDeclarations. It lead to an invalid list of actor_ids in open_scenario.py:

```python
actor_ids = []
for actor in sequence.iter("Actors"):
    for entity in actor.iter("EntityRef"):
        entity_name = entity.attrib.get('entityRef', None)
        for k, _ in enumerate(joint_actor_list):
            if joint_actor_list[k] and entity_name == joint_actor_list[k].attributes['role_name']:
                actor_ids.append(k)
                break
```

After these loops, actor_ids only contained the id/index of the None that is added further up int the code to the joint_actor_list which lead to the errors.

PR #755 (2021-04-20): Fixed OSC parser issue on Nonetype actor
Resolved exception "AttributeError: 'NoneType' object has no attribute
'attributes'" in OpenScenarioParser.

Fixes #753

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/scenario_runner/755)
closed 2021-06-15 by fabianoboril