carla-simulator/scenario_runner #756: Unable to run an external .xosc file in scenario runner

aakashsehgal, 2021-04-23:
**Describe our question or idea**
I have a simple .xosc (1.0) file generated by an external tool and also the corresponding .xodr file. To run this in Carla, I replace the "Town1" in the .xosc file with the path of my .xodr file and then try to run this scenario via 

`python scenario_runner.py --openscenario "C:\Tools\Carla\CARLA_0.9.11\scenario_runner-0.9.11\srunner\examples\NCAP\CPLA_SD_v1.xosc" --output`

This crashes my Carla server and i have the following error 

> WARNING:[SR:OpenScenarioConfiguration]: Wrong OpenDRIVE map in use. Forcing reload of CARLA world
> Traceback (most recent call last):
>   File "scenario_runner.py", line 600, in <module>
>     sys.exit(main())
>   File "scenario_runner.py", line 590, in main
>     result = scenario_runner.run()
>   File "scenario_runner.py", line 494, in run
>     result = self._run_openscenario()
>   File "scenario_runner.py", line 482, in _run_openscenario
>     config = OpenScenarioConfiguration(self._args.openscenario, self.client)
>   File "C:\Tools\Carla\CARLA_0.9.11\scenario_runner-0.9.11\srunner\scenarioconfigs\openscenario_configuration.py", line 59, in __init__
>     self._parse_openscenario_configuration()
>   File "C:\Tools\Carla\CARLA_0.9.11\scenario_runner-0.9.11\srunner\scenarioconfigs\openscenario_configuration.py", line 90, in _parse_openscenario_configuration
>     self._set_carla_town()
>   File "C:\Tools\Carla\CARLA_0.9.11\scenario_runner-0.9.11\srunner\scenarioconfigs\openscenario_configuration.py", line 175, in _set_carla_town
>     self.client.generate_opendrive_world(str(data))
> RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000

The scenario is extremely simple. How can i use the scenario runner to execute such scenarios ? Also i am using a binary version of Carla 0.9.11. Could this be a problem in importing the new .xodr files ?

**Desktop (please complete the following information):**
 - OS: Windows10
 - CARLA Version [0.9.11 - Release Package]
 - Python version [3.7]

fabianoboril, 2021-04-23:
Most likely the OpenDRIVE map is the root cause of this problem. Can you try to load it with CARLA's config.py tool (you can find it in PythonAPI/util).

aakashsehgal, 2021-04-23:
No, i cannot load the OpenDRIVE map using the config.py tool. As soon as i try to load it , the simulator crashes with the error 

> LowLevelFatalError [File:Unknown] [Line: 102] Exception Thrown: failed to find road elevation.

Do you think that the map is problematic ? SInce i can load the same map in other simulation tools.

glopezdiest, 2021-04-27:
Hey @aakashsehgal, can you check your memory usage when loading the map? We recently saw that the TrafficManager had a memory leak that caused some non CARLA maps to result in a timeout when trying to load them. [This PR](https://github.com/carla-simulator/carla/pull/3937) fixes it, in case that's the case

aakashsehgal, 2021-04-27:
HI @glopezdiest : i dont think so this is the problem. My CPU load is somewhere at 20%. As soon as i try to load the .xodr file via the config tool, my simulator crashes with the error mentioned above. Any other idea how to check if it really is a memory leak problem ? The map is nothing but a straight road. So nothing complex at all. Do you have a sample .xodr which i can compare with ?

glopezdiest, 2021-04-27:
Could you send me your *.xosc* and *.xodr* files? I'll see if they are failing for me

aakashsehgal, 2021-04-27:
sure...how shall i send them ? email ?

glopezdiest, 2021-04-27:
You can compress them into a ZIP and directly paste it in the comment

aakashsehgal, 2021-04-27:
[osc_carla.zip](https://github.com/carla-simulator/scenario_runner/files/6385636/osc_carla.zip)

glopezdiest, 2021-04-28:
Yeah, I'm seeing the error. The server is crushing from malloc size, which is causing the client to timeout. I'll investigate about it, see what's going on.

glopezdiest, 2021-04-28:
After checking your *xodr*, the problem is that your *elevationProfile* doesn't specifiy the elevation at `s=0`, so the server crashes. You can manually add it and it'll work just fine:
```
<elevationProfile>
    <elevation s="0.0000000000000000e+01" ..."/>         <!--  Add something like this -->
    <elevation s="2.0000000000000000e+01" ..."/>
    <elevation s="4.0000000000000000e+01" ..."/>
    <elevation s="6.0000000000000000e+01" ..."/>
    ...
</elevationProfile>
```

PD: superelevation also has that problem but we don't really parse it, so no need to change it

closed 2021-05-03 by aakashsehgal

bapukokare, 2022-06-02:
Error: Action pathfollowing trajectory not supported.
 please check bellow code  we getting some  error  while running externally generated  .xosc file  . initially map load properly  & vehicle spawn but action not happening. Get an error for Action pathfollowing trajectory not supported. 
                                   <PrivateAction>
                                     <RoutingAction>
                                        <FollowTrajectoryAction>
                                            <Trajectory name="S1_Ego_Trajectory" closed="false">
                                                <Shape>
                                                    <Polyline>
                                                        <Vertex time="0.0">
                                                            <Position>
                                                                <WorldPosition x="3.612" y="1.689" z="0.0"  h="0.0" p="-0.0" r="0.0" />
                                                            </Position>
                                                        </Vertex>
                                                        <Vertex time="0.2">
                                                            <Position>
                                                                <WorldPosition x="5.632" y="1.689" z="0.0" h="0.0" p="-0.0" r="0.0" />
                                                            </Position>
                                                        </Vertex>
                                                        
                                                    </Polyline>
                                                </Shape>
                                            </Trajectory>
                                            <TimeReference>
                                                <Timing domainAbsoluteRelative="absolute" scale="1" offset="0"/>
                                            </TimeReference>
                                            <TrajectoryFollowingMode followingMode="position"/>
                                        </FollowTrajectoryAction>
                                    </RoutingAction>
                                </PrivateAction>