carla-simulator/scenario_runner #757: Invalid syntax when testing the example

dimmo90, 2021-04-23:
I'm trying to run the test of scenario runner as per the documentation, however I am getting this error:

`dim@dm-OMEN-by-HP-Laptop-15-dh0xxx:~/scenario_runner$ python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld
Traceback (most recent call last):
  File "scenario_runner.py", line 37, in <module>
    from srunner.scenariomanager.scenario_manager import ScenarioManager
  File "/home/dim/scenario_runner/srunner/scenariomanager/scenario_manager.py", line 23, in <module>
    from srunner.scenariomanager.watchdog import Watchdog
  File "/home/dim/scenario_runner/srunner/scenariomanager/watchdog.py", line 14, in <module>
    import simple_watchdog_timer as swt
  File "/home/dim/.local/lib/python2.7/site-packages/simple_watchdog_timer.py", line 11
    def __init__(self, callback, check_interval_sec: float = 0.01, trigger_delta_sec: float = 1, identifier=None):
                                                   ^
SyntaxError: invalid syntax
`
I am using python 2 to run it as I am trying to link scenario runner with CARLA and rosbridge, but it seems that simple_watchdog_timer.py is written in python 3?

Any help or a work around to achieve this link would be appreciated.

 - Ubuntu 18.04.5
 - CARLA 0.9.11
 - Python 2.7.17
 - Scenario runner 0.9.11 [package release]
 - ROS kinetic

dimmo90, 2021-04-27:
All sorted now. Unfortunately I couldn't find a version that was compatible, however i found a workaround by replacing watchdog.py with a version from an earlier release (one that didn't reference simple_watch_timer).

thiagodeborba, 2021-05-02:
hey @dimmo90 , i am getting the same issue how did/can you replace the watchdog.py file, what was the workaround, can you elaborate more, please?

ParthAD3, 2021-05-02:
hey @dimmo90  i am facing the same issue, so how did you changed the file watchdog.py, what was the workaround, can you please elaborate more?

dimmo90, 2021-05-02:
Hi @ParthAD3,

What version of scenario runner are you running? 0.9.10? because that should have the correct file and I wouldn't expect this problem to occur. It happened to me because I was trying to run scenario runner 0.9.11 with carla 0.9.11.

1) go into scenario_runner/srunner/scenariomanager/watchdog.py. Check line 14. If it says "import simple_watchdog_timer as swt" instead of "from threading import Timer" then I think you might have the 0.9.11 version. Because they advise you to have the same version of scenario runner as you do carla, I suggest you download scenario_runner 0.9.10 and the problem should go away (hopefully)
2) if it turns out you do have scenario_runner 0.9.10, then please post the error here and I'll try to help you out.

Cheers

ParthAD3, 2021-05-02:
Hey, yes i was in tag 0.9.11 
but then i am gettig this problem 
[image]

ParthAD3, 2021-05-02:
hey @dimmo90 yes i was on tag 0.9.11,  i have changed to 0.9.10 but now i am thrown with this issue " Configuration for scenario FollowLeadingVehicle_1 cannot be found 
i have updated the python, carla, and scenario runner paths,
simulation is playing 
all files are there, can you tell me the issue now ?
[image]

ParthAD3, 2021-05-02:
hey @dimmo90 yes i was on tag 0.9.11, i have changed to 0.9.10, sry to bother on silly mistake

closed 2021-06-15 by fabianoboril