carla-simulator/scenario_runner #759: The actor is not spawning to execute a scenario test

ParthAD3, 2021-05-03:
**Describe the bug**
When i run this simple argument: 
python scenario_runner.py --scenario FollowLeadingVehicle_1 --output
it executes with:
Preparing scenario: FollowLeadingVehicle_1
ScenarioManager: Running scenario FollowVehicle

Then after when i run the python manual_control.py --filter <ego_vehicle>
pygame launches, i have my ego vehicle, But upon giving throttle i dont see the actor (the leading vehicle in this case)

sometime, when i drive around i see the actor being spawnned but then when i drive close by it does not move (seems dead)
and after collision (aswell no response) after sim time out, in the output i get this:
[image]

**Expected behavior**
upon giving throttle to the ego the actor(leading vehicle) should be launching and driving maintaining a distance 

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - CARLA Version [e.g. CARLA 0.9.10]
 - Python version [e.g. 2.7]
 - Version [e.g. 0.9.10]

glopezdiest, 2021-05-05:
Hey @ParthAD3, you seem to be running CARLA's manual control, not ScenarioRunner's, is that the case?

ParthAD3, 2021-05-05:
Hey, yes i am running carla's manula control, as when i tried to run manual control from the scenario runner i was thrown with the error, that there are no examples

glopezdiest, 2021-05-05:
Then, you are most likely missing the `$CARLA_ROOT/PythonAPI/carla` folder at your PYTHONPATH

ParthAD3, 2021-05-05:
So, the __init.py__ was not defined in my carla/PythonAPI/examples folder 

That's why it was not loading, it's working now, thanks for the quick answers

closed 2021-05-05 by ParthAD3