carla-simulator/scenario_runner #761: Scenario runner

Purvil2810, 2021-05-04:
> Hello Guys,

i am new at CARLA. I just downloaded carla release 0.9.11 and same for scenario runner. 
As per the Carla doc., i tried to open the scenario runner with the command: 
python scenario_runner.py but i am getting an error : 
‘ import carla 
No carla module found’

can anyone tell me why i am getting this error. What could be the problem?

glopezdiest, 2021-05-05:
Hey @Purvil2810, when using CARLA's Python API, you need to add several folders to the PYTHONPATH, have you done that? Here is a template:
```bash
# ${CARLA_ROOT} is the CARLA installation directory
# <VERSION> is the correct string for the Python version being used
# In a build from source, the .egg files may be in: ${CARLA_ROOT}/PythonAPI/dist/ instead of ${CARLA_ROOT}/PythonAPI
export CARLA_ROOT=/path/to/your/carla/installation
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla/dist/carla-<VERSION>.egg
```

Purvil2810, 2021-05-05:
My directory is like 
Carlafolder>pythonAPI >carla>dist>.egg file

is it correct?
If it is not then how can we set your recommended template in windows? 
please guide me.

glopezdiest, 2021-05-05:
I haven't set up environment variables in a long while on Windows but according to our [SR Docs](https://carla-scenariorunner.readthedocs.io/en/latest/getting_scenariorunner/),  the windows template is something like:
```bash
# %CARLA_ROOT% is the CARLA installation directory
# <VERSION> is the correct string for the Python version being used
# In a build from source, the .egg files may be in: ${CARLA_ROOT}/PythonAPI/dist/ instead of ${CARLA_ROOT}/PythonAPI
set CARLA_ROOT=\path\to\your\carla\installation
set PYTHONPATH=%PYTHONPATH%;%CARLA_ROOT%\PythonAPI\carla\dist\carla-<VERSION>.egg
```

closed 2021-11-23 by glopezdiest