carla-simulator/scenario_runner #766: XML library instead of string concatenation

AzaelCicero, 2021-06-14:
**Is your feature request related to a problem? Please describe.**
Currently srunner/scenariomanager/result_writer.py uses string concatenation to format XML with results. I propose to use XML library to this. Then for example we won't have to worry about security vulnerability i.e. https://owasp.org/www-project-top-ten/2017/A4_2017-XML_External_Entities_(XXE).

**Describe the solution you'd like**
Replace string concatenation with XML library. 

**Describe alternatives you've considered**
None

**Additional context**
None

fabianoboril, 2021-06-15:
Hi,
would you be able to prepare a draft PR for this?

AzaelCicero, 2021-06-15:
Yes, I would be able to prepare such PR if you think it will be viable addition to Scenario Runner.