carla-simulator/scenario_runner #775: Running Scenarios with Manual Control

ashepele, 2021-07-12:
**Describe our question or idea**
Hello Scenario_Runner developers! I'm pretty new to scenario_runner and am trying to reproduce the simple scenario that you show at the 1:08 minute mark in the [CARLA Talks 2020 - Introducing ScenarioRunner](https://www.youtube.com/watch?v=dcnnNJowqzM) video.

I'm issuing the following commands (as suggested at the 3:31 time mark in the same video)

`python3 scenario_runner.py --scenario DynamicObjectCrossing_1`
`python3 manual_control.py`

I'm finding that I'm getting 2 ego vehicles in different parts of the map. scenario_runner.py produces an ego vehicle and manual_control.py also produces its own separate ego vehicle.

**Expected behavior**
I would have expected that manual_control.py would be able to find the existing ego vehicle and apply manual control to it.

Insead, it looks like one option is to run the commands in the following order
`python3 manual_control.py`
`python3 scenario_runner.py --scenario DynamicObjectCrossing_1 --waitForEgo`

In this case, manual_control.py spawns the ego and then scenario_runner.py moves the ego to the start location of the scenario.

What is the correct way to do this? Is there a way to have manual_control.py find an existing ego vehicle and apply manual control to it?

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - CARLA Version [e.g. CARLA 0.9.10]
 - Python version [e.g. 3.6.9]
 - Scenario Runner Version [e.g. 0.9.10]

fabianoboril, 2021-07-13:
Hi,

did you use the manual_control.py of CARLA or the one next to scenario_runner.py? You have to use the latter.

ashepele, 2021-07-13:
Hi @fabianoboril , I was indeed using Carla's manual_control.py. Thank you for catching my mistake!

closed 2021-07-14 by fabianoboril

GeneralMak, 2021-07-27:
Hello,do you know how to lay out the scenario you created on the route? At the same time ,do you know how to create your own planned route in a town?