carla-simulator/scenario_runner #776: No module named 'examples.manual_control'

jbirtel, 2021-07-13:
Hi,

I just tried to run the test scenario.
I first had an issue that the PYTHONPATH for scenario_runner.py was not found. I use openpilot and somehow the PYTHONPATH is set to ...../openpilot and I can't find the file where it is saved (but actually not .profile and not .bashrc), even though I set the PYTHONPATH correctly in bashrc file.

I could somehow solve this by adding the paths with sys.path.append('......') to the python file (scenario_runner.py).

Afterwards, I was able to start scenario_runner and in CARLA, the weather changed. When I now tried to complete the test scenario by starting 'manual_control.py' in the scenario_runner folder, I got the same error, so again I added the paths manually by adding sys.path.append('.....'). This solved the error 'no module name carla'.

However, now I am stuck with the following error:

`File "manual_control.py", line 71, in <module>
    from examples.manual_control import (World, HUD, KeyboardControl, CameraManager, CollisionSensor, LaneInvasionSensor, GnssSensor, IMUSensor)
ModuleNotFoundError: No module named 'examples.manual_control'
`
By now, I added the following lines to manual_control.py: (/.../ is on purpose only for this post)

`import glob
import os
import sys
sys.path.append('carla/PythonAPI')
sys.path.append('/home/..../carla/PythonAPI/carla')
sys.path.append('/home/.../carla/PythonAPI/carla/agents')
sys.path.append('/home/.../carla/PythonAPI/carla/dist/carla-0.9.11-py3.6-linux-x86_64.egg')
sys.path.append('/home/.../carla/PythonAPI/examples')`

Can anybody please help me either finding where that PYTHONPATH variable is overwritten and resetting that or by adding another line for sys.path.append()? That'd be great!

glopezdiest, 2021-07-13:
The path you want is `/home/.../carla/PythonAPI`, which is your case, is the only one that doesn't have the `home/.../` initial statement. Is this also the case for your *manual_control.py* file?

jbirtel, 2021-07-13:
No, in my _manual_control.py_ this is indeed correct. Sorry, that I put that mistake in the description above. So all paths start with '/home/..../carla/', also that one with PythonAPI. So, I guess this is, unfortunately, not the problem here.

glopezdiest, 2021-07-14:
Then I don't know what your problem is. I just removed all PYTHONPATHs related to CARLA and did your `sys.path.append(...)` and it works fine for me. In any case, I don't think you should follow this route, as there are a lot of file where CARLA is imported and you're going to have to add it to all of them.

Jk7913, 2021-07-14:
Maybe try to add __init__.py file to the carla/PythonAPI/examples folder.

jbirtel, 2021-07-15:
@glopezdiest Hmm, unfortunately it does not work for me, there is still the error. I don't know why, but the PYTHONPATH approach does not work for me. That's why I tried the approach with the sys.append, and I was able to solve the first problem.

@Jk7913 thanks for your help, but this does not change anything (I simply added two empty files in carla/PythonAPI and carla/PythonAPI/examples

jbirtel, 2021-07-15:
If I run the following command:
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}PythonAPI:${CARLA_ROOT}PythonAPI/carla:${CARLA_ROOT}PythonAPI/examples:${CARLA_ROOT}PythonAPI/carla/dist/carla-0.9.11-py3.6-linux-x86_64.egg:${CARLA_ROOT}PythonAPI/carla/agents

before the execution of manual_control.py it worked for me.

closed 2024-06-14 by glopezdiest