carla-simulator/scenario_runner #777: OpenSCENARIO world position

Jk7913, 2021-07-14:
Hi all,

I'm experiencing some problems when placing actors in OpenScenario scenarios by "World position". I have created scenarios based on a OpenDrive file (Carla Demo xodr files ass well), and in that scenario actor is positioned by World coordinates. When running that scenario with ScenarioRunner, it seems that actors are then shifted on this map. Sometimes this mean few meters down the road, sometimes they cannot be spawned.
I also went the other way, if I try to run simple example scenario "PedestrianCrossingFront.xosc"  in Esmini, actors are shifted on the map. Similar thing in Town04.
[image]

 I have tried this on the other simulator as well and only CARLA is causing the problems for me. Placing actor by other types of position works fine. I'm only having a problem with world positioning and to me it seems like the origin of this coordinate system is somewhat shifted in ScenarioRunner. Any idea why this might be happening?

Thank you

 - OS: 18.04
 - CARLA Version 0.9.10.1
 - Python version 2.7
 - Version 9.10

fabianoboril, 2021-07-14:
Most likely, this is a left-hand vs. right-hand coordinate system issue.

If you use the CARLA coordinates (e.g. those that you get in manual control), you have to use a "CARLA:" at the start of the description tag
<FileHeader revMajor="1" revMinor="0" date="2020-03-24T12:00:00" description="CARLA:CyclistCrossing" author=""/>
If you use the OpenDRIVE coordinate frame, you have not to put the leading "CARLA:"

Does this help?

Jk7913, 2021-07-14:
Haven't done much testing, but for now it seems that that was the problem. Thank you

closed 2021-07-14 by Jk7913