carla-simulator/scenario_runner #778: Debugging Scenario_Runner py_trees

ashepele, 2021-07-15:
**Describe your question**
Hello Scenario_Runner developers! I have a question about the correct way to debug the py_trees that get created in the _create_behavior function of a scenario class.

For example, I'd like to create a modified version of the DynamicObjectCrossing scenario and visualize the py_tree at each update to make sure my scenario is running as expected.

Following the instructions here:
https://py-trees.readthedocs.io/en/devel/visualisation.html, I've added the following code to object_crash_vehicle.py

Add to import statements
```python
import functools
```

Add function above all classes

```python
def post_tick_handler(snapshot_visitor, behaviour_tree):
    """
    Prints an ascii tree with the current snapshot status.
    """
    print("\n" + py_trees.display.ascii_tree(behaviour_tree.root,
                                             snapshot_information=snapshot_visitor))
```
Before the return statement in DynamicObjectCrossing._create_behavior(). 

```python
        behaviour_tree = py_trees.trees.BehaviourTree(root)
        snapshot_visitor = py_trees.visitors.SnapshotVisitor()
        behaviour_tree.add_post_tick_handler(functools.partial(post_tick_handler, snapshot_visitor))
        behaviour_tree.visitors.append(snapshot_visitor)
```

**Expected behavior**
I would have expected that an ascii visualization of the py_tree would be printed at every update, but I'm not seeing anything being printed.

Using the command `py_trees.display.print_ascii_tree(root)` produces something like the output I would have expected.

```
OccludedObjectCrossing
[-] Sequence
    --> TransformSetterTS3walker
    --> TransformSetterTS3coca
    --> Braking
    --> TimeToArrival
    --> Braking
    (o) keep velocity
        --> walker velocity
        --> walker drive distance
    (o) keep velocity other
        --> walker crossing lane accelerate velocity
        --> walker move distance for lane crossing 
        --> ego vehicle passed prop
    --> walker stop
    --> Destroying walker
    --> Destroying Prop
    --> End condition ego drive distance
```

Could you please advise if this approach is recommended for debugging py_trees? If not, how would you recommend ensuring that the scenario execution flow is running correctly?

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - CARLA Version [e.g. CARLA 0.9.10]
 - Python version [e.g. 3.6.9]
 - Scenario Runner Version [e.g. 0.9.10]

fabianoboril, 2021-07-15:
Hi,
you can use the debug commandline option. This will print the ascii-formatted tree.

ashepele, 2021-07-15:
Thank you @fabianoboril !

I have a follow-up question

It looks like the debug output produces something like a function call stack instead of an actual tree visualization.

[image]

Do you know if it's possible to modify the output so that it looks more like the output of `py_trees.display.print_ascii_tree(root)`?

```
OccludedObjectCrossing
[-] Sequence
    --> TransformSetterTS3walker
    --> TransformSetterTS3coca
    --> Braking
    --> TimeToArrival
    --> Braking
    (o) keep velocity
        --> walker velocity
        --> walker drive distance
    (o) keep velocity other
        --> walker crossing lane accelerate velocity
        --> walker move distance for lane crossing 
        --> ego vehicle passed prop
    --> walker stop
    --> Destroying walker
    --> Destroying Prop
    --> End condition ego drive distance
```

fabianoboril, 2021-07-16:
Usually you should see the tree as well.  If this is not happening, maybe the tree is not properly constructed or there is a version issue? You can uncomment the following line to get a plot of the constructed tree:
https://github.com/carla-simulator/scenario_runner/blob/aaea1de86b98db4aae6bcd4d3c433f8e698e9c30/srunner/scenariomanager/scenario_manager.py#L111
Maybe this gives some indication.

glopezdiest, 2021-11-23:
Solved. Feel free to reopen it if more help is needed

closed 2021-11-23 by glopezdiest