carla-simulator/scenario_runner #779: carla.actor bounding_box.location influenced by WorldPosition argument "h" in xosc TeleportAction during Storyboard-Init

DennisKlpng, 2021-07-16:
**Describe the bug**
Hi folks, I have a rather weird issue while running xosc-files in the ScenarioRunner. I reproduced it with the usage of ManualControl from the Carla python examples and running the ScenarioRunner with the according xosc file.
When there is a teleport action during Init, the bounding_box.location of the vehicle apparently depends on (at least) the heading angle of the WorldPosition in the TeleportAction, with absolute implausible values as the end result (e.g. an y-position of 150 for the bicycle model in Carla). Reproduction seems to be easier with the bicycle vehicle model

**To Reproduce**
Steps to reproduce the behavior:
1. Adapt the scenario: i used the FollowLeadingVehicle.xosc as the starting point. I replaced the vehicle type of the adversary-vehicle (line 32: <Vehicle name="bicycle" vehicleCategory="bicycle">). In addition it is necessary to set a heading angle in the teleport action in line 98. 
2. Export the bounding box location. I added the following lines to the game_loop function of manual_control.py in PythonAPI\examples of Carla:
```
        carla_world = client.get_world()
        actors = carla_world.get_actors()
        for actor in actors:
            actor_type_id = str(actor.type_id)
            if actor_type_id.startswith("vehicle.diamondback"):
                print("actor_type_id:")
                print(actor_type_id)
                print("BB location:")
                print(actor.bounding_box.location)
```
3. Start the Scenario Runner with the scenario, start manual control
4. With a heading angle value of "0" in FollowLeadingVehicle.xosc (Line 98 from 1. ), the result is:
```
actor_type_id:
vehicle.diamondback.century
BB location:
Location(x=0.000020, y=0.000000, z=0.811402)
```
(as expected)

5. With a heading angle value of "3.02434" in FollowLeadingVehicle.xosc (Line 98 from 1. ), the result is:
```
actor_type_id:
vehicle.diamondback.century
BB location:
Location(x=0.000020, y=154.619476, z=0.811431)
```

**Expected behavior**
The location should be invariant to player position

**Desktop (please complete the following information):**
 - OS: Windows 10
 - CARLA Version 0.9.11 (WindowsNoEditor)
 - ScenarioRunner Version 0.9.10
 - Python version 3.7.9

**Additional context**
I noticed that the bounding box extent of the bicycle-model (vehicle.diamondback.century) doesn't seem to be plausible as well, since the y-extent is 0. Should be a Carla-issue though, isn't it?

fabianoboril, 2021-07-19:
> Location(x=0.000020, y=0.000000, z=0.811402) 
This seems wrong. The origin is elsewhere. Can you check if this is position is persistent over multiple frames. Most likely, the actor was not yet fully initialized by CARLA when you printed the bounding box paramters.

DennisKlpng, 2021-07-19:
Yes, the bb-location for the bicycle stays the same. Apparently the issue only occurs for the bicycle, not for pedestrians, for them the location is just x=0.000000, y=0.000000, z=0.000000 (even with the afromentioned heading > 0)

fabianoboril, 2021-07-19:
Adding @glopezdiest

glopezdiest, 2021-08-04:
Hey sorry for the delay, as we were releasing 0.9.12. I'll be talking a look at it, see if I can find what's happening