carla-simulator/scenario_runner #780: How to use Carla, ScenarioRunner & OpenPilot

jbirtel, 2021-07-22:
**Describe our question or idea**
I am trying to use the ScenarioRunner to simulate Scenarios in OpenScenario format. Additionally, I use OpenPilot. The problem occurs, when trying to combine them. Until now, I tried only the FollowLeadingVehicle scenario.
The ego vehicle which is spawned by OpenPilot is not recognized by the Scenario_Runner. I also tried to use the "--waitForEgo" Flag, but this does not change anything here. So how is it possible to specify, that the vehicle, that is spawned when starting OpenPilot, triggers the scenario, instead of the vehicle used right now. I hope you get my point? 

**Expected behavior**
I expect that I can combine OpenPilot and Carla+ScenarioRunner to be able to run scenarios with the given vehicle (of OpenPilot). 

**Desktop (please complete the following information):**
 - OS: Ubuntu 18 LTS
 - CARLA Version: 0.9.11
 - Python version: 3.6
 - Version 0.9.11
 - OpenPilot: 0.8.4

I am happy about anything, that can help to run the scenarios.

fabianoboril, 2021-07-23:
I am not familiar with OpenPilot, but in general you can do the following: Let OpenPilot spawn the vehicle with the rolename "hero", and use that name for the OpenSCENARIO object, i.e. <ScenarioObject name="hero">. Then use --waitForEgo.

jbirtel, 2021-07-29:
Sorry for the late reply, I tried to figure out where I can replace the name of the vehicle in the OpenPilot files, but I didn't find something yet. How is it generally possible to define the name of the vehicle in python ? Maybe it is already sufficient to simply add that to the file that starts OpenPilot.

jbirtel, 2021-07-29:
I added the line `blueprint.set_attribute('role_name', 'hero')` in the bridge.py file before the vehicle actor is spawned, so here https://github.com/commaai/openpilot/blob/d7ab611bb0a616374e28eb4f0cdf46bb0172bb4b/tools/sim/bridge.py#L198 basically.

I then started the bridge.py of OpenPilot with the addition that I specified town Town01 as this was needed, scenario runner like `python3 --scenario FollowLeadingVehicle_1 --waitForEgo` and also the manual_control as usual. However, the consoles with the scenario_runner both kept printing "waiting for ego vehicle"  or if I changed the starting order, OpenPilot was unavailable. Do I need to make additional steps?

fabianoboril, 2021-07-29:
The line that you have linked is for the camera not for the vehicle. Is just the link wrong, or did you add the rolename to the wrong actor (camera not vehicle)?

jbirtel, 2021-07-29:
Damn it! I saw that there are different actors for camera and vehicle, but I somehow mixed up the lines in the vim editor. After inserting `vehicle_bp.set_attribute('role_name','hero')` before this line, so in the [correct positon](https://github.com/commaai/openpilot/blob/d7ab611bb0a616374e28eb4f0cdf46bb0172bb4b/tools/sim/bridge.py#L175), it worked and the scenario could be run. So thank you really much for your help!

closed 2021-07-29 by fabianoboril

PatrickFSLin, 2022-05-12:
@jbirtel 
Hi, I am trying to reproduce the results you did

Carla: 0.9.12
ScenarioRunner: v0.9.12
OpenPilot: 0.8.14

I have done connect carla, scenariorunner, and openpilot, but the question when I manually drive ego_vehicle or start ACC, NO other vehicle spawned... it supposed to spawn vehicle in front of ego in the scenario "following_leading_vehicle"

jbirtel, 2022-05-12:
So the steps that I did were
1. Start Carla and play the scene (so any map) in there, as for usual scenes
2. Start OpenPilot with the specified map of the scenario. You should get into another mode (in the docker) where you can start ACC, stop ACC, stop the bridge.py, manually drive the vehicle.
3. You have to insert `vehicle_bp.set_attribute('role_name','hero')` in the bridge.py with a vim editor or so, while you are still in that special mode. This must be inserted after the variable vehicle_bp is declared. You need to save that and exit the editor.
4. Here I am actually unsure about the correct order out of the box , but I think you first need to start the bridge.py with the correct town (—town <name>) and scenario runner with the waitForEgo flag. Then, on starting ACC, some milliseconds afterwards the leading vehicle is spawned.

These were the steps that worked for me at least.

PatrickFSLin, 2022-05-12:
@jbirtel 

Thank you SO MUCH!!

Finally I figure out why I cannot see the spawned vehicle after I ACC my ego_vehicle. The key point is the ORDER.
I did run scenario_runner first and then bridge.py.

Thanks again!