carla-simulator/scenario_runner #782: How to lay out the scenario you created on the route ? How to create your own planned route in a town

GeneralMak, 2021-07-27:
Hello Scenario_Runner developers!  In the 6:00 minute mark of the CARLA Talks 2020 - Introducing ScenarioRunner video,you show an example of setting up scenarios on the route.Can you talk about how to set up a scenario on the route,and at the same time,how to plan a route in a town? Thank you !

glopezdiest, 2021-08-04:
@GeneralMak, sure thing :slightly_smiling_face:.

The first thing you need to do is to create your own python scenario, in the same way that it is done for scenarios outside routes.

Then, again, same as before, you need a config file to describe where the scenario is placed. For routes, this is a *json* and not an *xml*. [Here](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/data/all_towns_traffic_scenarios.json) is our example, but do understand that this englobes all possible scenarios for all town, so it quite a big file. At the end of the day, it is just a dictionary that filters the town, then the scenario, and then gets its data. `"transform"` is the position of the ego, and instaed of spawning an ego there, it waits for the ego to pass though that position to start the scenario. `"scenario_type"` can be anything you want, and it'll be mapped to the real scenario name [here](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/scenarios/route_scenario.py#L53).

Next, you need to create your route (check [this example](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/data/routes_training.xml)). There's no need to do it with a lot of precision, just add the key waypoints to the xml and SR will automatically find the shortest route between them.

Lastly, you have to run the [RouteScenario](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/scenarios/route_scenario.py), which will create the route, get all the scenarios that pass through that route (randomly picking one if there are several very close-by), and trigger the scenarios you added when needed. It also adds several criteria. 

In order to run an agent with routes, you'll have to create something like [this](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/autoagents/npc_agent.py) and run

`python scenario_runner.py --route <path_to_routes_file> <path_to_scenarios_file> <route_id> --agent <path_to_agent_file>

The *route_id* is the id of the route, in case your file has several routes and you only want to run one of those.

glopezdiest, 2021-11-23:
Closing this issue as this seems to be solved. Feel free to reopen it if needed

closed 2021-11-23 by glopezdiest