carla-simulator/scenario_runner #793: Scenario window closes within split seconds

werewolfdev, 2021-08-30:
I am trying to run a  simple scenario created by running the python script created, mostly a subset of FollowLeadingVehicle scenario. everything loads properly but the screen closes within split seconds. Iam using manual control
**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - CARLA Version [e.g. CARLA 0.9.10.1]
 - Python version [e.g. 3.7]
 - Version [e.g. 0.9.10]
Any ideas would be helpful.

werewolfdev, 2021-08-30:
I dont have NVIDIA Graphics card and I am running on my Intel Graphics. Can you let me know if this could be causing this issue. The example scenarios also crash and close sometimes

glopezdiest, 2021-08-31:
Hey @werewolfdev, could you provide more information about how are you running the scenarios? There are several arguments that you can use which might be affecting your result

werewolfdev, 2021-08-31:
Hey @glopezdiest . Firstly I open the World window running the Carla shell script. Then I run the config XML file of the scenario from the scenario runner folder and eventually run the python script corresponding to the scenario and then open the manual_control python script . The scenario opens at first and then closes within 30 seconds or so every time

glopezdiest, 2021-08-31:
Okay, several questions then:
1) When you run the scenario, does a message similar to "Running Scenario *name of the scenario*" appear?
2) What are you doing with the manual control? Are you moving through the lane? Whenever you first move, the scenario should start
3) Scenarios have a timeout of around 1 minute, so if nothing is done after that time, they are automatically shut down. Is it possible that this is what's happenning to you?

werewolfdev, 2021-09-01:
1. Yes the message do appear
2. iam actually performing a functionality similar to follow leading vehicle. The ego vehicle has to only follow the leading vehicle going at a constant velocity without collisions . Whenever i start the ego vehicle the new actor car is spawned in front and within split seconds of motion the window closes
3. I have set timeout to 360 seconds and have seen the output window for the scenario simulation. It mostly runs for 20 seconds or so before closing.
I have seen in many places to always go for NVIDIA Graphics for smooth usage of CARLA. As i dont ahve one , Iam having doubts if that is leading to these crashes. initially my world window used to close automatically but once i turned quality level to low it stopped that problem. Kindly let me know your views @glopezdiest. thanks for your time

werewolfdev, 2021-09-06:
Any updates @glopezdiest . Thanks

glopezdiest, 2021-09-08:
Oh sorry for the delay, I didn't see the message. I'm sorry to say that if it is anything related to graphics, then I probably won't be able to help as I haven't heard about that before. 

From Scenario Runner sides, everything seems to be working fine, you are creating and executing the scenario correctly. Could you send me the files you are using to execute it? I'll give it a try

werewolfdev, 2021-09-08:
kein Problem. Will get a new graphics cards of NVIDIA soon . Will update on this ticket once I retry . Thanks a lot.

werewolfdev, 2021-09-20:
@glopezdiest  Thanks for the support .NVIDIA Driver helped .Closing this issue

closed 2021-09-20 by werewolfdev